dekuNukem/Nintendo_Switch_Reverse_Engineering #2: Rumble data

lyra833, 2017-03-15:
Hey! Thank you so much for the packet data! I'm working on getting it into a library so there's joycon access for non-Switch stuff like VR games and the like. I wasn't able to get my hands on any extra joycons, so you breaking one out and doing this has been a lifesaver!

I was wondering if you've made any headway in figuring out how HD rumble data is sent over, especially when the update request just looks like a fixed value and doesn't seem to transmit any information to the joycon. Changing any bytes in the update request just seems to not get a response from the JC I tested.

Once again, thanks a ton!

dekuNukem, 2017-03-15:
Thanks for the attention! Come to think of it I haven't really captured any data with vibrations in them, I'll do one today or tomorrow and see what's different.

lyra833, 2017-03-16:
Awesome, thanks!

closed 2017-03-16 by lyra833

dekuNukem, 2017-03-16:
I just updated a capture file with rumble in it. Looks like the console sends a longer command to the joycon to start a rumble, I haven't looked at the details yet but it's out there.