dekuNukem/Nintendo_Switch_Reverse_Engineering #7: HID Protocol for Bluetooth / USB

octo-byte, 2017-03-15:
Hi, 

Loving the work you've done - thanks ever so much. Couldn't find you on social media, so am just gonna drop a message here. 

Have you been able to sniff the USB or Bluetooth HID protocol it's using? Been trying to bodge a driver together, and don't really have the tools to analyse the protocol. Whilst the buttons and joysticks work fine, I believe the console sends a feature report to the joy-cons to activate gyroscope input. So far I've not had much luck getting the gyroscope and accelerometers to send anything over the standard protocol. 

Could be me, or it could be Nintendo's very kind way of introducing security through obscurity.
Some help on the matter would be great if you are free. 

Thanks
mail [at] octo [dot] im

alejandrorangel, 2017-03-15:
I'm also working to reverse engineer the joycons, I found this documentation. I think that Nintendo might use something similar to the joycons.

[http://wiibrew.org/wiki/Wiimote](Wiimote HID)

dekuNukem, 2017-03-15:
thanks, I did figure out the accelerometer and gyroscope data in the joycon status packet today, but only in the physical connection bewteen the joycon and console. I haven't began working on the bluetooth level yet, but I'll keep it in mind when I do.

forerofore, 2017-03-16:
just to make you guys aware, there is a program similar to glovepie called freepie, the possibility of developing a module for the joycons may be faster/easier than an entire new driver. thanks and gj with all of this.

octo-byte, 2017-03-16:
@alejandrorangel Cool, thanks for that. I might try and throw some of the Wiimote packets at it this weekend and see if it reacts to any of them.

aspalmer, 2017-03-16:
Over USB the pro controller just does a handshake, then switches to bluetooth.
Descriptor:
[procontroller descriptor.txt](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/files/848574/procontroller.descriptor.txt)
Beagle analyzer trace:
[data-center-windows-x86_64-v6.72.zip](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/files/848567/data-center-windows-x86_64-v6.72.zip)

octo-byte, 2017-03-21:
@aspalmer That actually makes sense from the HID reports I was getting back, actually. Was able to get the HID info from the USB device, although no data was being sent. However when I paired the USB and Bluetooth together some test code, I noticed one thing. 
Over the USB, the HID protocol reports that there are 2 extra components called "Z-Acceleration, and Z-Rotation". I can only imagine this is the gyroscope and accelerometer. However, as it was over USB, no data was sent. This leads me to believe you do indeed have to send a feature report packet to the controller via bluetooth to enable the gyro/acc.

rlaferla, 2017-03-23:
I'd love to see an iOS API for the JoyCon via Bluetooth.

riking, 2017-03-25:
I have some notes on the Bluetooth hid buttons. The left stick & right stick buttons are adjacent button numbers. So are + - and home, capture.

It sends the stick data as hat input, though. (0-8, 8 is neutral, 0 is up) Annoying.

mfosse, 2017-03-26:
@riking What are you using to capture the Bluetooth traffic?

riking, 2017-03-27:
@mfosse Oh, I just straight up connected the joycon to my Linux laptop and read the input data with `hidapi`. So it's default drivers.

Was about to try sending some of the 0x19 packets as OUPTUT or FEATURE reports and see what happens.

riking, 2017-03-27:
Yup, I found the stick data.

```
uint8_t *pckt = buf65 + 2; // skip the HID header (0x21) and checksum byte
// print buttons...
uint8_t *stick_data;
if (jc->left_right == 1) {
    stick_data = pckt + 4;
} else {
    stick_data = pckt + 7;
}
uint8_t stick_unk = stick_data[0]; // maybe checksum
// horizontal stick is nibble swapped, or maybe bit swapped
uint8_t stick_hz = ((stick_data[1] & 0x0F) << 4) | ((stick_data[1] & 0xF0) >> 4);
uint8_t stick_vert = stick_data[2];
printf("Stick %c: [%02X] %d %d\n", L_OR_R(jc->left_right), stick_unk, -128 + (int)(unsigned int)stick_hz, -128 + (int)(unsigned int)stick_vert);
```

riking, 2017-03-28:
Oooookay, looks like I'm going to need to pair the controllers before I move further - they started connecting to someone else's console, which is erroring out my Bluetooth stack.

Here's what I have so far:
https://github.com/riking/joycon/blob/master/src/joycon_input.c
https://github.com/riking/joycon/blob/master/src/joycon.h

octo-byte, 2017-03-28:
Can confirm the joy-cons and pro controller do operate using the standard HID protocol, all except the gyro and accelerometer.

riking, 2017-03-28:
I've got a working "combine both joycons into a single controller device" program here: https://github.com/riking/joycon

mfosse, 2017-04-03:
I'm working on a vJoy feeder for the JoyCons here: https://github.com/mfosse/JoyCon-Driver

shinyquagsire23, 2017-04-08:
The Charging Joy-Con grip seems to have two interfaces with an in/out endpoint per interface for each Joy-Con. I haven't managed to get any replies from input I've sent, *however* the Joy-Con grip, once an HID session is started, seems to send packets matching the first MAC address handshake command for UART every time the Joy-Con are inserted and ejected. The format seems to go as follows:

``` 81 01 <03 for eject, 00 otherwise> <inserted Joy-Con type, 01 for Left, 02 for right> <MAC>```

This can also be observed from Wireshark:
[image]

The charging grip also has the same STM32 chip as the pro controller, and the same 5-pin layout found on the dock. Maybe the firmware can be dumped same as the dock to learn more about UART and Bluetooth commands @dekuNukem?

[image]

shinyquagsire23, 2017-04-08:
Checked out the USB capture from @aspalmer and sending 0x80 0x01 gives me that MAC packet. It seems to respond to some of the other output reports in the capture similarly, so it seems that two Joy-Con in a charging grip effectively functions as a Pro Controller with two interfaces.

lyra833, 2017-04-09:
I've been wrangling around with the Joy-Con and think I've figured out how the rumble data is sent over, starting from byte 13 in the packet sent from the Switch. (It doesn't look like standard HID...)

Byte 13 is a constant, `0x10`. It's probably an identifier, as it's always `0xf` for all pings *not* containing rumble data. Byte 14 looks like some sort of timekeeper in the event of the packets colliding; it starts at `0xa`, increments with every ping sent until `0xf`, then loops back around to `0x0`, `0x1`, etc. Changing either of these bits doesn't trigger any rumble at all. There are then 8 bits, 6 of which seem to correspond to the 6 faces of the Joy-Con; but my drivers started crapping out before I could dope out the last 2. The first of the 8 (Byte 15) seems to be some sort of global intensity modifier; as changing it seemed to jolt the whole thing more or less, so that leaves 6 faces, one global modifier, and ?????.

I've managed to trigger some rumble events, but my laptop drivers are suddenly refusing to play nice with the Joy-Cons. I'm not sure if it's hardware failure or something to do with manipulating this data, but I'd love to have some fresh eyes verify it. I verified it against @DekuNukem's BotW data and it seems to check out with respects to which bits change when during the bomb rumble. (I would love to try it with [this](https://www.youtube.com/watch?v=hZQd5rBbPNQ), because it would let us isolate which bits correspond to which edge of the Joy-Con.)

Once someone else verifies, we can add it to the README?

Oh, and as a clarification because my Bluetooth has been hell, they're sending the same data over BT as they are over serial, right? The only thing that's different is the clock speed?

shinyquagsire23, 2017-04-09:
I have the firmware for the Joy-Con charging grip, doing a quick search for A1 A2 A3 A4 gets me a hit so it does seem that they talk to Joy-Con and have some sort of shim to HID. The firmware is pretty small though, 0x10000 bytes total.

riking, 2017-04-09:
@shinyquagsire23 When the Bluetooth stack of the joycon and your computer desync (or whatever that is), just do a physical pair with your Switch, long press the sync button, and connect again from your computer.

dekuNukem, 2017-04-09:
The top one is probably RESET, if I remember correctly. I tried to dump the pro controller firmware once but for some reason the st-link can't seem to find the chip. Anyway looks like they used STM32 for handling USB communications.

mfosse, 2017-04-09:
Has anyone managed to read any gyroscope or accelerometer data over bluetooth? I haven't had any luck so far.

mfosse, 2017-04-09:
@riking I figured out what stick_unknown is
it appears that the X component of the stick data isn't just nibble reversed, specifically, the second nibble of second byte is combined with the first nibble of the first byte to get the correct X stick value:

`uint8_t stick_horizontal = ((stick_data[1] & 0x0F) << 4) | ((stick_data[0] & 0xF0) >> 4);`

I don't have any idea what the other nibbles are, they seem random to me, but I haven't looked at them too closely.

shinyquagsire23, 2017-04-10:
It seems there is indeed a set of HID commands which allow sending UART commands, however they cannot be accessed until handshaking is complete, and I'm not sure of the command to do that (seems to be done elsewhere from the main loop).

Commands which seem to exist at 0x0800BB00 are:
```
80 01 (Handled elsewhere? Returns MAC packet)
80 02 (Do two handshakes 19 01 03 07 00 91 10 00, 19 01 03 0B 00 91 12 04)
80 03 (Do baud switch)
80 04 (Set something to 1)
80 05 (Set something to 0)
80 06 (Something with sending post-handshake command 01 00 00 00 00 00 00 00 01 06 00 00, maybe baud related as well?)
80 91 ... (Send pre-handshake command? Has some weird lookup table stuff)
80 92 ... (Something with pre-handshake commands?)
01 ... (Send post-handshake command? Sends 0x31-large UART starting with 19 01 03 07 00 92 00 00 with checksum edited in and the rest of the HID request pinned on)
10 ... (Same as above)
```

riking, 2017-04-11:
I left my controllers on the grip for a few days and now packet[1] is 0x6E instead of 0x8E. Battery level?

riking, 2017-04-11:
Here's what the Linux Kernel gives for the report descriptor:

https://www.irccloud.com/pastebin/P60ntzyH/

@shinyquagsire23 can you dump `/sys/kernel/debug/hid/_____/rdesc` with the Charging Grip?

shinyquagsire23, 2017-04-11:
@riking That file doesn't seem to exist for me, but the charging grip also doesn't function as a controller by default.

Also, I've managed to get a full initialization of the Joy-Con via the charging grip. The charging grip is able to send any UART command, so I am able to pull for full input at around 16ms per input, in addition to being able to dump my Joy-Con SPI firmware over HID. My HID program can be found at my repo, https://github.com/shinyquagsire23/HID-Joy-Con-Whispering

Hopefully with that, it should be possible to test Joy-Con functionality more in-depth. I did try to have the same HID commands sent over Bluetooth but it seems it's not the same protocol, unfortunately...

riking, 2017-04-11:
Sorry, the underscores weren't meant to be literal - replace them with the kernel-assigned device identifier.

> but it seems it's not the same protocol, unfortunately...

Dang :(

shinyquagsire23, 2017-04-11:
@riking Ah, I did replace the underscores but I guess I forgot to escape something and it failed, https://gist.github.com/shinyquagsire23/89ea38220e221950a233cd23f2fde28f

riking, 2017-04-11:
Okay, wow. Much more straightforward than the Bluetooth dump. Looks like a charging grip is almost all you need to use the joycons as a controller, I'll have to get one and add button mapping support in to my program.

Still doesn't help with the Bluetooth protocol except to reinforce that yes, 0x80 is definitely a bridge to the wires and it's definitely not available over Bluetooth.

octo-byte, 2017-04-11:
Bit of a long shot, but input 33 and 129 look like pitch/yaw inputs of a gyro

riking, 2017-04-12:
33 is the status report (0x21) sent in reply to a 0x1 HID output report and parsed here https://github.com/riking/joycon/blob/master/src/joycon_input.c#L53

mfosse, 2017-04-13:
Do we have any way to capture bluetooth traffic between the JoyCons and the switch? Or would we need something like this?: https://www.adafruit.com/product/2269 If I get one I'll be sure to document what I find here

riking, 2017-04-13:
It's not BLE, it's standard Bluetooth. Which means you need to pull a full
MITM. The hardware for that is in the range of $20,000USD according to
##ubertooth on Freenode.

mfosse, 2017-04-13:
I was afraid that might be the case, but wasn't sure, I'll probably pick up a charging grip sometime soon and see if I can find anything.

riking, 2017-04-15:
Got a charging grip, but I'm going to continue the Bluetooth research.

Based on the data from the other Joy-Con I just tested with, the battery data is actually just a single nibble - `2` was low according to the console, so probably a 0-8 or 1-8 scale.

mfosse, 2017-04-15:
I also just got a charging grip, If I find anything I'll post it sometime soon.

dekuNukem, 2017-04-15:
just a reminder guys, feel free to submit a pull request with findings so they'll have more visibility.

Kekskruemel, 2017-04-17:
Don't know if this help you guys but matlo, creator of gimx programmed a little bluetooth proxy here is the link https://github.com/matlo/l2cap_proxy greetings

riking, 2017-04-17:
Looks like the Switch is rejecting my attempts to connect to it from my computer - guess I need a BT radio that supports MAC spoofing.

mfosse, 2017-04-19:
According to https://github.com/matlo/l2cap_proxy,
"The master and the device have to be paired with the bluetooth dongle."
I'm pretty sure there isn't a way to get the switch to become discoverable, much less pair with anything but the joycons, I wasn't able to get anything to work, either; I also tried a different bluetooth mitm/proxy I found here: https://github.com/conorpp/btproxy. I might get an ubertooth one, and see if I can make any progress with that.

mfosse, 2017-04-20:
I've bought an Ubertooth One, and with some scanning I've managed to get the device "Nintendo Switch" to show up with an associated BTADDR, hopefully I'll make some more progress later today when I get the chance, progress might be a bit slow as I familiarize myself with the Ubertooth One. If anyone wants me to run any tests or needs any specific info, just let me know.

Also, can anyone @shinyquagsire23 @riking confirm whether or not the JoyCons w/charging grip or pro controller communicate over USB when connected to the dock (or do they just charge)? They are clearly capable, so I wouldn't be surprised if they do; If they do, it should be relatively easy to sniff/intercept and it could provide some insight on the bluetooth JoyCon <-> Switch communication.

ojbr, 2017-04-20:
@mfosse Limited understanding of how it works on the Switch side of things, but I wasn't able to get any HID input via USB. Only via Bluetooth.

finger563, 2017-05-17:
I'm currently trying to write some code to just communicate to a single JoyCon using an L2CAP Bluetooth socket, and can get it to respond to some commands, but the responses seem meaningless to me and I can't get it to send me input data, have you all figured out how to properly get it to send data just over Bluetooth on a BT socket?

elandyboys92, 2017-06-11:
Hi, im not a programmer, but does anyone here know a way to control the nintendo switch using a pc and a generic controller. Either having the switch recognizing the input as a procontroller or as joycons?

dekuNukem, 2017-06-11:
I made a custom board for remotely controlling joycons https://github.com/dekuNukem/joyAnalog

You can take apart a pair of joycons, solder wires on to button test points and connect them to the board. It's a lot of work but before the joycon is completely reverse engineered this is probably the only way. Here s a video: https://www.youtube.com/watch?v=aj1JLARaZQg

elandyboys92, 2017-06-12:
Ok, so I have no idea how to do that :S Would you sell me one? If I want to
play on the switch I need this because I have a special controller (im
quadraplegic c6) that allows me to play. I can use my controller with a pc
easily and can play on the wii u, xbox and ps4 with workarounds, but cant
currently play the switch. I would really apreciate if you could send me
one of those circuit boards and maybe some instructions on which points
need to be soldered.

Check out my controller if you want :v
https://www.youtube.com/watch?v=h9dm2prcSL0

elandyboys92, 2017-06-14:
> watch?v=h9dm2prcSL0

Hi DekuNukem, would you please sell me one of those integrated circuits you
made?? I cant play he switch any other way D:

AUmrysh, 2017-06-14:
elandsboy92, I would be happy to help you by soldering a pair of joycons. Do you have specs for your controller, an interface or pinout or something?

Kekskruemel, 2017-06-15:
I am also interested. Is there any layout for the board ?

ReconDaemon, 2017-06-15:
It looks like the following BOM cost, ignoring shipping, would be the price to build 3 of the boards (since the minimum order is 3 PCBs):
3x PCB from OSH Park - $7.15
3x STM32F072C8T6 from Mouser - $9.45
6x ADG714 from Mouser - $29.52
3x AT24C08D (EEPROM) from Mouser - $1.08
6x MBR120 Schottky diode from Mouser - $2.22
3x sot23-5 voltage regulator - $0.36
3x USB Micro-B connectors from Sparkfun - $4.50 (they're currently sold out, might be able to source elsewhere)
6x KMR2 switch from Mouser - $2.10
the other resistors, caps, and jelly beans maybe $10 or something

Considering all of that, the boards are approx $20-$25 each. Might be able to do a group buy to bring the cost down. I'm not sure what resistor and capacitor values you need.

elandyboys92, 2017-06-15:
AUmrysh My controller is pretty flexible, you can check the user manual here: https://docs.google.com/document/d/1I4GtzR9qVmfWkX-bSEiWwPzeTTwI0vBQqb-P3fBXsk4/edit#heading=h.f2fylzjkqgrc

Basically you can connect it to pc, wii, wiiu, xbox360, xboxone, ps2, ps3, and ps4 using the usb, and/or different adapters (like cable adapter and cronusmax that i do have)

I can program buttons to be held, turboed or switch them from position. Heres a picture of the back of the controller:
[image]

I also have this: 
[image]

Which is the pinout for the expansion port, but I made a db25 cable and checked for continuity and the programming I make in the controller wont work, so that would be like my last resort.

You can check more details of my controller here:
http://www.broadenedhorizons.com/ultimate-arcade2
https://www.youtube.com/watch?v=h9dm2prcSL0
https://www.youtube.com/watch?v=iQznFavtekw&t=4s

Ideally, I´d connect it via usb to the switch or connect it to the pc and have the pc send info to the soldered joycons. If you have questions you can email me at elandyboys92@gmail.com

Thanks!

elandyboys92, 2017-06-15:
ReconDaemon I would easily pay those $25+ shipping/labor if someone is interested :v

elandyboys92, 2017-06-15:
AUmrysh Oh, I also have this file with the values of an HIDTest i did on my controller
[UA2.txt](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/files/1079272/UA2.txt)

shinyquagsire23, 2017-06-20:
@aspalmer Since 3.0 supports Pro Controller 100% wired, I'm curious if you'd be able to do another USB MITM to see what it requests. I have the protocol down but I can't say I'm not curious whether about I missed something or not.

Kekskruemel, 2017-06-20:
Does someone have a transparent usb proxy ? 

https://github.com/matlo/serialusb Doesnt work with Switch Update 3.0 because the controller doesnt register via usb on a pc to start the usb proxy.

Looks like he needs to talk via bt to the switch before switching to usb mode.

popeter45, 2017-06-22:
does anybody know if it would be emulate a joycon / pro controller using a raspberry pi zero in gadget mode?
either that or using as pi 3 over bluetooth?

shinyquagsire23, 2017-06-22:
@popeter45 Should be doable using GadgetFS/FunctionFS since you can define your USB descriptors, though I dunno if they speak HID directly.

KevinCathcart, 2017-06-23:
I would like to point out the following project, which uses an expanded version of the pokken controller descriptors to allow for analog sticks, and access to all buttons. This could prove useful to anybody trying to create custom wired controllers for the Switch, as long as they don't need HD rumble, NFC, or gyros: https://github.com/progmem/Switch-Fightstick

Not particularly useful for the other main use cases of this reverse engineering ( like custom bluetooth controllers, or analog/gyro on a pic from the joycons), but thought we probably want to note this here since this seems to be the most complete repository of reverse engineering information.

popeter45, 2017-06-23:
@elandyboys92 i too would be interested in one of these boards albeit i live in uk so shiping would be ££

[153 more replies]