dekuNukem/Nintendo_Switch_Reverse_Engineering #17: Firmware injection?

ivo708, 2017-04-24:
I dont know much about programing or anything but, if the firmware can be dumped,could be then ijected the firmware(of a lower version), so it works like a downgrade?
Or can only be injected on the same console?
Thats just my opinion and i dont know if this can be done so...
-Thanks

joushou, 2017-04-25:
Why are you interested in "downgrading your *joycons*?

ivo708, 2017-04-26:
Docking station firmware dump

The docking station uses a STM32F048 microcontroller. It's actually labeled as STM32P048 because it uses the FASTROM option where ST pre-programs the flash memory inside the factory. It has 32KB flash memory and 6KB RAM, runs at 48MHz.

It uses SWD debugging and programming interface, and interestingly the programming testpoints are on the PCB and clearly labeled. After connecting a ST-Link programmer to it reveals that the chip is not read-protected at all, so a firmware dump was easily made. I'm not going to post it in the repo, but if you want it just ask.

I mean this, isnt talking about the switch firmware?

magnusnordlander, 2017-04-26:
No, that's talking about the firmware of the docking station.

CTCaer, 2017-09-20:
It's now doable.