dekuNukem/Nintendo_Switch_Reverse_Engineering #19: It might be interesting

SleepyBrett, 2017-05-31:
To dig into the hdmi dock and figure out why standard usb-c -> hdmi dongles don't work.

claymullis, 2017-06-08:
Are you curious as to why the Switch itself can't output using to HDMI using its USB-C port? I don't know the specifics, although I think the white papers for the dock mention some video processing and HDMI support. 

It may be a feature stripped from hardware but, you're right, USB to HDMI is a ubiquitous standard now that even works pretty well on most phones. That makes me think they just didn't write a driver the switch's USB port to output whatever it is that HDMI needs. Probably to get people to use the dock in order to charge their switch, and to encourage use of it's multiple form factors.

lordkitsuna, 2017-07-22:
I am more curious if this could be used to create a driver to trick the Switch into thinking a computer was a dock. Would be really great for streaming if we could just pull the video from right over the USB with 3.1 or something.

magnusnordlander, 2017-07-24:
That's not really how USB-C alt modes work.

fox8091, 2017-11-06:
From what I remember, it's actually USB-C to DisplayPort, then DisplayPort to HDMI.

LinkEscapist, 2017-11-08:
@fox8091 you are right,according to the tear down,it is a DP to HDMI thing,some one I know that works for  semi manufacturer said that the way nintendo do it is not "ordinary",maybe some non structured VDM thing is involved

LinkEscapist, 2017-11-08:
Could you please try the DP adapters?will they work?If not, I believe nintendo played some tricks

SleepyBrett, 2017-11-20:
I have a usbc->dp adapter at home, I'll give it a swing.