dekuNukem/Nintendo_Switch_Reverse_Engineering #32: Questions about Joycon power and charging

AntoineTurmel, 2017-08-30:
It's 5V but how many amperes ?
Does the Switch's Joy-con rail constantly provide the power ?

CTCaer, 2017-08-30:
Depends on the circumstances. Docked? Undocked? Charging grip?
Normally they charge at around 125mA to 500mA.

Maximum supported by TI BQ24072 charger chip inside the Joy-Con is 1.5A.
This same chip is responsible of cutting off the charging and supply to the battery.

Lastly, the joy-con can't operate with power through the rails. Like phones, they are always powered by the battery.

AntoineTurmel, 2017-08-31:
When a Joy-Con is on the Switch Rail, Undocked, with Switch charging or 100%.
I was wondering about powering something else than a Joy-Con using the Switch rail.
Thanks for your reply :)

Sur3, 2017-09-22:
So the Joy Cons have an internal charging control, does that mean I could directly connect the 5V Pin and a GND Pin to the corresponding Pins of an USB Cable for tethered charging while playing?!
Are the GND Pins all connected together early in the circuit, so it would be enough to connect only one of the GND Pins for charging?

CTCaer, 2017-09-22:
@Sur3 
Yes. It's what the charging stations do.
About GND, probably. Test with a multimeter or just try them.

@AntoineTurmel
I am little bit late to answer. :P
Unfortunately I didn't make notes and I don't want to desolder anything for now.

AntoineTurmel, 2019-01-10:
I was wondering does the Switch deliver 5V constantly even if there are no Joy-Cons connected ?
I did a test (printed a 3D part based of RCM jig) with Pin 1 and 4 (from Switch) and my multi-meter show less than 1V instead of 5V, is that normal ?

AntoineTurmel, 2019-02-02:
Ok so I figured out reading the documentation, I should have done this carefully, so I assume I need to fake an handshake first and then report the battery status: https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_subcommands_notes.md#subcommand-0x50-get-regulated-voltage

CTCaer, 2019-02-02:
Switch decides when to charge, yeah.
And has different percents for Dock and Handheld mode.
What you can do is either use a custom joycon fw or mitm switch' HID sysmodule

AntoineTurmel, 2019-02-02:
Yeah, I think from the meantime, I'll use a battery or rely on the USB-C of the console to make my device ;)

nick-s5, 2022-10-09:
> Yeah, I think from the meantime, I'll use a battery or rely on the USB-C of the console to make my device ;)

What happened with your project? I’m working on a custom Joy-Con grip and I want to add USB charging to it. I’m not very experienced with circuits, so I was hoping I could simply connect USB ground and 5V to pins 1 and 4 respectively. Can’t find any more info on doing this and very few cases of people trying to do it, hoping you were successful with it and have some insight :) Were you able to make that work?

Thanks :)