dekuNukem/Nintendo_Switch_Reverse_Engineering #45: USB Spi Read

greasysock, 2017-10-25:
I'm trying to read from SPI via USB, but all I ever am returned is "FF" of the length of the request.

`81 92 00 92 00 31 00 00 6d 55 21 60 81 00 80 00 aa 97 77 18 18 81 00 90 10 26 80 00 00 0f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff`

Here is an example request.

Does anybody know what I should be doing differently? I'm using the subcommand 10 to read from spi

CTCaer, 2017-10-26:
Can you give an example of your full command?

greasysock, 2017-10-26:
Thank you for the thoughtful response and insight.

I did not realize that I had not calibrated this device before. However, even trying to read the serial for the controller I am given the same response.

here is a better look at what I am sending and receiving for the serial number.

`Send: 80 92 00 31 00 00 00 00 01 04 00 01 40 40 00 01 40 40 10 02 60 00 00 0e
Send: 80 92 00 31 00 00 00 00 01 05 00 01 40 40 00 01 40 40 10 02 60 00 00 0e `

`Receive: 81 92 00 92 00 31 00 00 bc 77 21 6a 81 00 80 00 b6 67 7a 10 78 7f 00 90 10 02 60 00 00 0e ff ff ff ff ff ff ff ff ff ff ff ff ff ff`

CTCaer, 2017-10-26:
Pro controller maybe? Pro controllers do not have S/N and they are 0xFF.
Additionally, your command and the reply structure are absolutely correct.

I would suggest to first dump the whole SPI of your device and then check if your individual spi read commands work and you got the correct data.

Also, read the documentation here to know what to expect and what not in each SPI address.

Lastly, if you want to experiment with USB commands other than x80, 
I can provide you the full list (not only what you can see here).

greasysock, 2017-10-27:
Yes, it is a pro controller. I would really appreciate seeing the full command list for USB.

greasysock, 2017-10-27:
Asking for the device model gave me the right response from SPI.

CTCaer, 2018-03-04:
@Davidobot 
What do you send and what do you get?

Davidobot, 2018-03-04:
Doing `ReadSPI(0x80, 0x12, 9);` to read from 0x8012 and get the user stick calibration data would hang the code for a while and then just output a completely zero array.
I use the same implementation of the spi read function as shown further up in this thread.

CTCaer, 2018-03-04:
The hanging comes from the retries. That's because in USB mode the x30 input report mode is enabled by default. So the code expects a x21 but also gets x30 reports. This can be mitigated by rejecting the x31 report around 8 times and then retry to send the command again.

To do this, you need to change the joycon_send_subcommand() to identify the x10 subcmd and check for x21 inputreports in hid_read(). And better change the hid_read to hid_read_timeout with 100ms timeout.

About the xFF problem, as I said before, this section is empty (all 0xFF) if the controller was never manually calibrated. 
You said in the email that everything works as it should in BT mode, so I assume you have calibration data there. Then what happens is that you reach **max_read_count**.

So, if max_read_count is the problem, it will probably be fixed if you implement the x30 reports reject mechanism above.

Davidobot, 2018-03-04:
Thanks for your help! Setting the input mode to 0x3F and then back to 0x30 solved the problem!