dekuNukem/Nintendo_Switch_Reverse_Engineering #54: The Joystick

ThatPersonOverThere, 2018-03-15:
I am new at this whole tinkering thing right now. I want to know if its possible to take the joystick & its connector (from the board) and port it to an entirely different board, (I get the plastic connector won't be easy to remove... heck probably impossible). What I hope to be able to do is make a keyboard matrix  on a different circuit board using a atmega32u4 micro controller (used in an Arduino pro micro). The keyboard will have 60 (10 rows x 6 columns) buttons, (idea based off: [https://medium.com/@monkeytypewritr/building-your-own-keyboard-from-scratch-bd0638c40850),](https://medium.com/@monkeytypewritr/building-your-own-keyboard-from-scratch-bd0638c40850)  but I plan to keep the joystick separate from the matrix and still connected to the micro controller.  The board will be powered by a micro usb connected to the usb input of a raspberry pi zero w (I won't use the gpio pins because I need to use a small screen that takes them up the way it will be connected if I can't get it to work through mini HDMI.).  The matrix acts as a keyboard, while the joystick acts like a mouse with a click. The reason I want the joystick from the joy con is because it is very low profile, and clicks. Since the ps vita, nor the psp joystick click they are virtually useless to me.  I need to better understand if the joystick has any special programming quarks to it, the map out of the pin connectors and a good way to connect them to the micro controller (if it needs resistors on the board etc)... I also want to know how to remove the rubber grip on the joystick (I don't want to break it) . The programming I can probably figure out its the hardware I'm not use to. I came to this place because it had the most information on the joycon and its joystick than anywhere else.  This is also very theoretical at this point, no schematics exists because like I said I'm new to this tinkering thing.

ThatPersonOverThere, 2018-03-24:
For anyone who wants to see the joystick's insides
https://imgur.com/t/joycon/sxIab 
Lots of useful pictures here.

TerminalSaint, 2018-04-14:
I'm tinkering with a switch joystick myself, with plans of adding it to a flight stick for additional analog control axes. I worked out the pin-out from the traces inside the stick.
Bottom view:
https://i.imgur.com/QZCymI7.jpg

I got it to work on a PC by hooking it up to a Teensy and using a sketch intended for a standard gamepad thumbstick.

The jankiest part is the connector. The conductor pitch on the FPC is about .5mm, too close to comfortably solder leads to. I desoldered the plastic connector from the board, carefully craved traces with an x-acto into a piece of scrap PCB, then soldered the connector to the PCB and secured it with epoxy. The PCB gave me large enough pads to comfortably solder wire leads. The whole process was a gigantic pain and there must be some better way.
https://i.imgur.com/myLUfFw.jpg

Hope any of that helps.

ThatPersonOverThere, 2018-04-14:
Thank you so much! That's really helpful!

A better way to get the connector off might be to bake the joycon board in the oven, haven't tried it personally yet (I was given that advice by a friend). I'll try to see if I can find the FPC connector for sale in reasonable selling portions, ie. 5 - 10 pieces rather than over a 1000, because what average consumer need that many (yes I have had luck finding parts on some sites, crazy enough on my search for tiny LCDs).  An alternative method to baking is the heat gun if you have one (I don't), and if you are really lucky you might be able to get away with using heated tweezers, or fine tipped soldering iron.

Literally as I wrote this, found some possible connectors, though I'm not sure they will do:
-- [https://nl.aliexpress.com/item/10-Pcs-FPC-FFC-0-5mm-Pitch-5-Pin-Drawer-Type-Ribbon-Flat-Connector-Top-Contact/32728524729.html](https://nl.aliexpress.com/item/10-Pcs-FPC-FFC-0-5mm-Pitch-5-Pin-Drawer-Type-Ribbon-Flat-Connector-Top-Contact/32728524729.html) 
also its in Dutch, I think.
-- [https://www.molex.com/molex/products/datasheet.jsp?part=active/0512810594_FFC_FPC_CONNECTORS.xml](https://www.molex.com/molex/products/datasheet.jsp?part=active/0512810594_FFC_FPC_CONNECTORS.xml) 
you might want to see if you can get a sample

As for a way to fix the 0.5mm pitch connections
--You might be able to etch a board to a finer point with ferric chloride and a fine tip sharpie. If you have access to a 3D printer you can tape a marker on the head and let it drag the marker on the board for precise lines. If the printer is good enough you might be able to get plastic to stick to the board, but its because the copper will be cold it might not stick. No printer, tape and a caliper work to draw lines. 
--Possibly if could try to take the HD rumble motor's very fine wire, heat the end and quickly attach the connector, but probably not the best solution considering its still difficult and the length if fairly short. Nor do I know the resistance of the wire and how well it would work as a connection... etc. 
--You might be able to take something _like_ a micro usb board as shown [https://www.ebay.co.uk/itm/Micro-USB-Socket-PCB-Board-Port-2-54mm-Pitch-Arduino-Chassis-Breadboard-Plate-/112649098776](https://www.ebay.co.uk/itm/Micro-USB-Socket-PCB-Board-Port-2-54mm-Pitch-Arduino-Chassis-Breadboard-Plate-/112649098776) 
and just ditch the micro usb and attach the connector, another long shot, I would probably use soldering paste rather than wire. The good thing is you can make soldering paste templates easily with tape and a ruler or clear plastic paper. If you don't want to do that take a small sewing needle and use that to place the paste.
--Or try this [https://www.adafruit.com/product/1325](https://www.adafruit.com/product/1325)

Hope this helps too!
(P.S. you did a really good job with the PCB you made, its super impressive)

TerminalSaint, 2018-04-16:
Some good thoughts there. I was being a bit impatient, I just went with what I had on hand to see if I could get it to work. I wish I'd seen that adafruit board, I just placed an order with them a couple days ago.

gamelaster, 2018-05-22:
Hello,
Thank you guys for awesome research. I want too use the Switch's thumbstick in my project, @ThatPersonOverThere did you found a good FPC connector for it?

TerminalSaint, 2018-05-22:
I found that a molex 54550-0571 works perfectly, though it still requires very fine soldering. I had some scrap PCB and ferric chloride kicking around so I etched up a small board for it.

I also managed to find some pretty cheap extra joysticks to experiment on by looking for OEM versions on AliExpress.

One thing I did notice once I had a reliable connection is that the thumbstick is not a standard 10K; it has a slightly reduced range. Windows controller calibration is able to obviate the issue, but I'm not sure how it might affect other applications.

gamelaster, 2018-05-22:
@TerminalSaint thank you, good to know. I found the cheapest thumbstick for switch for 5 USD (+ free shipping) on Aliexpress:
https://www.aliexpress.com/item/3D-Button-Analog-Joystick-High-Quality-Controller-Stick-Thumbstick-Replace-For-Nintend-Switch-Joy-Con-NS/32876226173.html