dekuNukem/Nintendo_Switch_Reverse_Engineering #59: Joycon Joystick Connector?

GingerOfOz, 2018-05-07:
I was wondering if anyone has a measurement of the width of the joystick FFC connector so that I could find a replacement connector that could make using Switch joysticks on other projects easier than trying to solder to that tiny ribbon cable.  I've ordered a couple from China but if I could get a breakout board designed and ordered before they get here that would be fantastic.  Thanks!

hattesen, 2019-01-06:
Reviving an old thread with some information.

I am considering using this joystick for an HMI controller that will be used for industrial/medical equipment, located next to a 1.8" color TFT display.
I have located several (Chinese) sources for 0.5 mm pitch 5-pin FPC connectors, similar to the one used on the Switch Joy-Con.
I include a couple of links to Aliexpress, but they are frequently short-lived, so you may choose to just search Aliexpress using this query: `fpc connector 0.5 mm 5p`

The FPC connectors are either "drawer" or "flip-top"-type, and the contacts either at the bottom (towards the PCB) or top.

The cost for 10 pcs 5 pin connectors is around $1.50 (inclusive of shipping).

If only I could locate some proper dimensional drawings of the thumbstick, I'd be really happy ;)

- [10pcs-FFC-FPC-connector-0-5-mm-4-Pin-5-6-7-8-10-12-14](https://www.aliexpress.com/item/10pcs-FFC-FPC-connector-0-5-mm-4-Pin-5-6-7-8-10-12-14/32886552793.html)
- [10Pcs/lots FPC Connector socket FFC 0.5MM Clamshell Ribbon Flat Bottom Contact Type 4/5/6/7/8/10/12/14/16/18/20/22/24/26/28/30p](https://www.aliexpress.com/item/10Pcs-lots-FPC-Connector-socket-FFC-0-5MM-Clamshell-Ribbon-Flat-Bottom-Contact-Type-4-5/32838332276.html)
- [10Pcs FPC Connector socket FFC 0.5MM Drawer Bottom Contact type 4/5/6/8/10/12/14/16/18/20/24/26/28/30/32/34/36/40/45/50/54/60P](https://www.aliexpress.com/item/10Pcs-FPC-FFC-Connector-cable-socket-0-5MM-16Pin-Drawer-type-Bottom-Contact/32728912605.html)

AlexFundorin, 2019-06-13:
I found a 2.54 breakout option but only for 6 pin. Already bought it. We'll see, is I could use it with 5pin cable.
https://nl.aliexpress.com/item/FPC-FFC-platte-kabel-overdracht-plaat-is-rechtstreeks-DIY-0-5mm-afstand-connector-6-P-8/33006390185.html?spm=a2g0s.9042311.0.0.66be4c4dLjoA3C

RVeldhuyzen, 2019-11-12:
@AlexFundorin How did the breakout bords work out?

AlexFundorin, 2019-11-12:
@RVeldhuyzen It works. I soldered it wrong the first time, but after resoldering everything is fine now.

RVeldhuyzen, 2019-11-12:
@AlexFundorin That's good to hear. Thanks for the info. I'll order a few for my current project :)

nekuneko, 2020-10-31:
I figured out the Joycon FPC connector listed on the fist image is:
Molex 503480-0500 / 503480-0540

Nevertheless, the most popular replace part for this connector on Aliexpress is this one:
[image]

You can buy it from this [link](https://es.aliexpress.com/item/4000277430226.html).
This part looks like to be:
Omron XF2W-0515-1A
- 0.5mm pitch 
- 5 Contacts
- Dual Contact

This connector looks like to be discontinued at Omron's website.

claytonsulby, 2021-01-26:
I'm looking to use the connector as well. I've found another Omron with the same specs as above but isn't obsolete. Is this an adequate replacement for the above part?

https://www.mouser.com/ProductDetail/Omron-Electronics/XF3M-0515-1B/?qs=vLWxofP3U2xVspNW6OQpUA%3D%3D

nekuneko, 2021-01-29:
> 
> 
> I'm looking to use the connector as well. I've found another Omron with the same specs as above but isn't obsolete. Is this an adequate replacement for the above part?
> 
> https://www.mouser.com/ProductDetail/Omron-Electronics/XF3M-0515-1B/?qs=vLWxofP3U2xVspNW6OQpUA%3D%3D

Beware @claytonsulby XF3M-0515-1B Samacsys footprint is completely different from XF2W-0515-1A part, its about two times bigger, but it could be a footprint error. If you want to repair your joycon try this part https://www.mouser.es/ProductDetail/538-503480-0500/ . But if you finally buy a couple of XF3M please send pictures to compare if there are the differences between connectors :)

Here I attached you a picture to you can see how it looks the molex part on my custom BLE controller. Joycon is fully working with this part. These FFC connectors are double sided so take a look to the pinout if needed.

[image]

PcMiguel22, 2022-02-05:
Si es te interés , pero no encuentro proveedor.
https://es.dhgate.com/product/smart-home-control-3d-analog-joystick-thumb/730729262.html#dspm=wapes.seo-onsh.list.24.&seo_type=1|#s-24-1;onsh|1584762918
[image]
[image]
[image]

gardenfrog99, 2023-02-03:
https://4layertech.com/products/joystick-breakout-board-pack-of-2