dekuNukem/Nintendo_Switch_Reverse_Engineering #62: Relapsing button switches using test points

HanssonS, 2018-07-11:
I have a project in mind that would require me to replace the switches for the dpad and face buttons. Soldering to the test points shouldn't be any major issue, but from the readme I gather that the way button presses are dealt with is a bit unconventional:
> Nintendo didn't use the traditional "one side pulled-up other side to ground" way of reading buttons, instead they used a keypad configuration where buttons are arranged in rows and columns.

Is this something I will have to take into account or is still a matter of bridging test points to ground?

dekuNukem, 2018-07-11:
Bridging test points to GND doesn't work in this case, you need to connect the button test points to the COL test point for it to register as a press. You can find COL test point in here:

https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/joycon_pcb.pdf

You will need some analog switches if you want to automate it. I have already made a control board to do just that: https://github.com/dekuNukem/joyAnalog

HanssonS, 2018-07-11:
Well damn, and here I was hoping this wouldn't become overly complicated. Any idea if it might be viable to remove the button stickers and solder to the points bellow?

dekuNukem, 2018-07-12:
It is certainly one way of doing it, but I think soldering to test points is easier since they are already there.

HanssonS, 2018-07-12:
Oh. I just realised what the issue is about. Electronics is a hobby of mine but it doesn't come naturally, so I apologise for being a bit slow on the uptake.

Going back to your previous reply, what exactly do you mean by 'automate', and why would I need analog switches? Is the resistance different depending on the buttons being pressed?

dekuNukem, 2018-07-13:
The buttons in joycon is arranged like this:

[image]

For simplicity only 4 buttons are shown, but this is true for every button on the joycon (apart from the stick button).

When you press a button, it connects to the COL signal and the processor register it as pressed. 

Now if we want to simulate a button press by soldering to the test point, we need to to connect the other end to COL signal for it to work. If you're replacing buttons, you can connect one end to the button test point, the other end to the COL test point, and it will work.

Analog switch is a lot like a regular switch, only you can control the switching with a electric signal instead of by hand. I used them to simulate button presses electrically by connecting and disconnecting the button to COL. It might be out of the scope of what you're trying to do, so don't worry too much about it.

HanssonS, 2018-07-13:
In that case, it sounds like there wont be any problem connecting each individual test point to COL with the switches of my choosing(?). 

Maybe I was a bit too vague explaining what I want to achieve; I'm designing new shells for the joycons and I don't like having the analog sticks vertically aligned with the face buttons. Since the left and right joycon aren't symmetrical it isn't really viable to simply move the analog sticks in relation to their respective the PCBs. Instead I was thinking of replacing the buttons outright, giving me more flexibility to change the layout. Maybe I'm getting water over my head, but then again, at this point I do have a pretty good track record.

dekuNukem, 2018-07-13:
yeah, I'm pretty sure that you can solder your new switches one end to the button test point, and the other end to COL test point, and it will work. Pretty straightforward.

HanssonS, 2018-07-13:
Thanks for the help :)

moribellamy, 2018-12-03:
Thanks for this info. Circuits noob here, so any help is appreciated:

Would it be possible to emulate a button press using RasPi GPIO along with a transistor? Something like turning the GPIO high could tell the transistor to connect the signal?

dimityrivanov, 2024-03-01:
@dekuNukem what about the potX and potY any idea how I can read they're input ?