dekuNukem/Nintendo_Switch_Reverse_Engineering #63: USB HID Motion/Gyro

WhileE, 2018-08-03:
So I've been reading through a ton of this repo, and it's amazing BTW, and I keep reading what seems like conflicting information.

Can anyone outright confirm/deny that joy-cons with a charging grip will communicate over USB instead of BT, and will this include gyro data if initialized properly?

I ask because I currently have an HID device (16u2)  plugged in via USB that emulates a HORIPAD controller, and it works great, but I'm looking at the possibility of incorporating gyro controls, but am unsure of the support over USB. I'm in the process of acquiring a charging grip to do some digging myself, but didn't want to waste the time if someone else can deny that this will ever work.

I see some saying you have to pair via BT then switch to USB, then there's others stating that the joy-cons only communicate over BT unless physically connected to the sides of the switch.

CTCaer, 2018-08-04:
Joy-con can communicate via BT and USB when in grips
You instruct them how. If already BT connected they respond to both. 
If USB first, you have to instruct them to switch to BT. Otherwise it's only USB.

That's why it seems conflicted ^

WhileE, 2018-08-06:
So when connected to the Switch, does it automatically tell them to switch to BT, and that's why the USB COM stops after a few seconds? I'm coming from the other side, I'm not looking to talk TO the joy-con, more emulate them. So have the charging grip, but have yet to understand how/if I can force USB only to glean some extra insight on the COM.

CTCaer, 2018-08-06:
> If USB first, you have to instruct them to switch to BT. Otherwise it's only USB.

So the usb comm does not stop. **Except** if you don't reply correctly.

WhileE, 2018-08-06:
I'm not the one replying, it's the Switch itself. As I said, I'm not talking TO the device, I'm just listening on the USB comm between the cons and the Switch, and it stops after a few seconds. So I have to assume the Switch basically tells it to stop USB comm, and does everything over BT.  
  
Digging around the settings I noticed there's an option to force the Pro controller to comm over USB only, but nothing for the joy-cons I could see.

CTCaer, 2018-08-06:
I was going to talk about that option.
You enabled it?

WhileE, 2018-08-06:
Yes, I did enable it, and power cycled. However when I connect the charging grip to the switch, nothing happens until I click some buttons, then it spits out USB comm for a few seconds, pairs, and then stops USB comm altogether.  
  
At first I thought it was because I was using USBProxy on the BBB to snoop the comm, but then I just my HORIPAD USB controller and it works just fine over USB.

mfosse, 2018-08-17:
@WhileE I might work on this eventually, since it's on my to-do list for a project I'm working on

I looked into this a while back & someone made an attempt but didn't get very far & it looks like they gave up, I'll link it & the stuff  I found on this when I have more time/get home

riking, 2018-08-26:
You can't really stop them from negotiating Bluetooth communication, kinda like a TLS handshake. You can do pure USB JoyCon <-> Custom, and pure USB Custom <-> Switch, but you can't force pure USB JoyCon <-> Switch.

Poohl, 2021-04-16:
Does it even switch to BT in Flight mode? because the Pro controller with a wired c-to-c connection works just fine in flight mode?