dekuNukem/Nintendo_Switch_Reverse_Engineering #69: Controller stick interface (5-wire flexible PCB)

hattesen, 2019-01-01:
Do you have any information of the pinout (and interface) of the 5-wire flexible PCB that connects the control stick with the controller PCB?

I would guess something like:

- GND
- VCC
- X (analog)
- Y (analog)
- Switch (short to GND)

patriot1889, 2019-01-03:
[image]

I found this on a different website.

closed 2019-01-03 by hattesen

benjaminaigner, 2020-05-12:
Do you maybe know if the X/Y axis are done via a resistor divider for up/down and left/right?
If yes & if my Ali order finally arrives, I can publish an Arduino Joycon Stick to Mouse interface :-)

THX @all for all the specs in this repository!