dekuNukem/Nintendo_Switch_Reverse_Engineering #71: LCD Pinout

spectrumm, 2019-01-15:
Has anyone reversed the lcd? I'm working on a custom project, and i bought a LCD, same as in the switch, to use with a raspberry, if possible id need the pinouts to connect it

spectrumm, 2019-01-26:
I cant seem to find any HDMI to DSI boards?

tpimh, 2019-09-23:
I have found this display: [E555HBM2](http://www.panelook.com/E555HBM2_EDO_5.5_OLED_overview_25401.html). Dimensions seem pretty similar to Switch original LCD, the connector is 30pin MIPI DSI and it is OLED. I wonder if it can be useful for your project.

tpimh, 2019-09-23:
Oops, just learned that I was wrong about the dimensions. Switch uses 6.2" screen, not 5.5" (Switch Lite uses 5.5").

ArcadeHustle, 2019-10-11:
They claim you can't trick Switch Lite into connecting to a TV.... I bet we can. https://www.cnet.com/news/you-cant-trick-nintendo-switch-lite-into-connecting-to-a-tv/ 

Perhaps need an FPGA to help, but I bet it can be done

ArcadeHustle, 2019-10-11:
@spectrumm did you see this https://hackaday.com/2014/08/19/a-mipi-dsi-display-shieldhdmi-adapter/
[image]

RetepV, 2020-04-21:
> @spectrumm did you see this https://hackaday.com/2014/08/19/a-mipi-dsi-display-shieldhdmi-adapter/
> [image]

No need for an fpga. This thing can do that hands down and probably cheaper.

https://www.analog.com/media/en/technical-documentation/data-sheets/ADV7533.pdf

andreacampanella, 2020-04-22:
No, that's a mipi to hdmi, it's used to add hdmi output to a host device with mipi, you need the other way around.

RockoBravo, 2021-02-09:
There should be a solution that doesn't require additional power.  It should be possible to jump of the LCD pins. Has anyone found a pin out?

maartenvb040476, 2021-07-26:
i am looking to make the lcd of the switch or switch lite work on the raspberry pi
so i made this schema and i know it can work like this but it is possible i am wrong about the data lines
https://easyeda.com/editor#id=85452c47e15d47ccb713e906936381b2
maybe someone with the right equipment kan recorded the signal from a switch

spookeo, 2022-04-11:
Has anyone achieved this?

tanNhungPhe, 2026-01-05:
anyone have information ???