dekuNukem/Nintendo_Switch_Reverse_Engineering #72: Soldering to the joycon

Spazzlo, 2019-01-21:
I'm trying to replicate the setup you have, but I just can't get the wires to solder to the contacts. Could you explain how you did it?

moribellamy, 2019-01-30:
Hey, I've struggle with this quite a bit as a newbie when it comes to circuits. Check out the ##electronics IRC channel, lots of people there have been helpful.

You should probably be more specific regarding your issues, as there is a lot that can go wrong. Ideally you will have a well regulated soldering iron. If you use too much heat, or you are not quick enough in your work, you might give too much thermal stress to the copper pads on the joycon and they will pop right off, and you'll be out $40.

This is the technique I've had the most luck with:

1) Use thin rosin-core 60/40 solder. Make sure your PCB is clean of dust (rubbing alcohol and a qtip can help)
2) Apply a small blob of solder to a well tinned tip.
3) Transfer the solder blob to a pad. Pressing the hot tip to the copper pad will heat it up enough in a few seconds. When the pad is hot enough, the solder will stick to it (you will see it gently move from the tip to the pad). Now you have a blob of solder on the pad.
4) Apply some flux to a stripped, multi stranded, small gauge copper wire. I just dip it in my flux.
5) With one hand holding the wire, and the other hand holding the soldering iron, gently press the tip of the fluxxed wire in to the solder blob. You will see the solder wick in to the strands.

There are lots of videos on soldering technique out there, just google for them. The above is considered bad practice, they always say you should apply the solder and the tip simultaneously to the joint, while holding the joined material in place with some kind of jig. I can't make that work here though, since the pads are too small.

Aezore, 2019-01-30:
Just to add that as a ballpark number, for small sized pcb you should work with any classic soldering iron (cheaper than regulated ones) around 20 watts if you are not used to soldering such small stuff to avoid burning things out inadvertedly. Also, a small flat ballpen sized tip will ease things out than the standard round tip as the flat point helps spreading heat better if you can get it.