dekuNukem/Nintendo_Switch_Reverse_Engineering #74: totally new left joycon PCB

moribellamy, 2019-02-26:
hey all,

Don't know if this is related to #73, but i just opened up a left joycon from an open box deal at best buy. for the first time, it was different than the one diagrammed here!

Did some digging and found this: https://www.ign.com/articles/2018/04/30/nintendo-working-on-new-redesign-of-joy-con-for-switch. Apparently, the left joycon has long had connectivity issues, so nintendo is redesigning it.

So you may want to try to buy older ones to use as much information from this repo as possible.

moribellamy, 2019-02-26:
heres a pic

[image]

CTCaer, 2019-02-26:
You had a pic here, why you removed it?
It was valuable.

It is indeed a redesign. Antenna is now at the bottom instead at top protruding.

- The antenna is signal separated with plated through holes. It now uses a different design.
- The baseband is now Cypress branded.
  (Broadcom bt part was owned by Cypress, even back then, Nintendo had the time to source millions of these.)
- It seems that IMU and SPI chip are now under the board
- Finally the new revision is called JCL-M-`11` instead of JCL-M-`01`

>So you may want to try to buy older ones to use as much information from this repo as possible.

That's partially  incorrect. The only thing that someone needs to do, is find the new pad positions, nothing else. And that's, if probing.

moribellamy, 2019-02-26:
thanks for the update @CTCaer.

i've uploaded the image again here. the first round i forgot to delete all the EXIF metadata :).

[image]

vic1707, 2020-07-29:
Hi if you want i got some of the soldering point for the buttons (sorry for the picture quality i don't have a microscope)
[image]
[image]

dimityrivanov, 2024-03-01:
@vic1707  is it possible using these pads with external MCU to press the buttons ?

vic1707, 2024-03-04:
I don't know, I haven't touched this for a while 😅