dekuNukem/Nintendo_Switch_Reverse_Engineering #75: home button on right joycon

moribellamy, 2019-02-27:
just wanted to share...if you bridge this test point to COL, it seems to register as a press on the "HOME" button.

[image]

Soul-Miku, 2022-11-07:
very excellent find, just one question, in the same photo where can you find it to be able to activate the bluetooth of the right joycon? Just like you did with the HOME button, what happens is that the FLEX connector of the SL/SR broke and the bluetooth button stopped working to pair the controls on the switch and on the PC, which I use a lot on the latter, I would like to know in which part of the board I have to connect the point, thank you very much if you answer me

CarbonatedMilk06, 2024-01-02:
Any way to contiunuity test the homebutton? Do you which test pads to use?