dekuNukem/Nintendo_Switch_Reverse_Engineering #81: Joy-Con HID Input Reports on ESP32

JSnowden33, 2019-06-07:
I'm trying to use BTstack on ESP32 to read basic data (buttons, joystick, and IMU) from a Joy-Con controller. BTstack has limited support for HID host applications, but the HID documentation here is great, so I've been trying to implement what I think is correct protocol. I am able to connect a Joy-Con to the ESP32 as an HID device by using the existing ["hid_host_demo"](https://github.com/bluekitchen/btstack/blob/master/example/hid_host_demo.c) from BTstack. I can send any output report I want - LEDs change accordingly so I know output reports work. 

However, the only input reports I ever receive are the "3F" standard input reports that contain button and joystick hat data. If I change the reporting mode using a subcommand, or change the name of the ESP32 to "Nintendo Switch" so the Joy-Con uses full reporting mode by default, no data comes in on the ESP32, not even in response to output reports. As far as I can tell, the ESP32 isn't receiving data (aside from 3F reports) on the HCI or L2CAP layers when it definitely should be. Is it possible that there is some incompatibility in BTstack that prevents it from receiving specific reports from a Joy-Con? Perhaps more L2CAP channels need to be opened somehow (there's only one for control and one for interrupt currently)?

I can post code if needed but the only modifications I've made to ["hid_host_demo"](https://github.com/bluekitchen/btstack/blob/master/example/hid_host_demo.c) are for output reports which don't seem to have any issue.

nathanRamaNoodles, 2020-05-09:
I'm having this exact same issue as well.  I checked the packets in Wireshark on my Windows computer, and it seemed to work fine with changing input reports.  So, I think the issue is in BTstack.

JSnowden33, 2020-05-09:
My issue was opening the L2CAP channel with too small an MTU. Using 64 instead of 48 works, so larger packets can come through. You can see an implementation here:
https://github.com/JSnowden33/Wii-Bluetooth-Replacement/blob/master/Software/ESP32/Wii_Bluetooth_Replacement/main/connect.c

closed 2020-05-09 by JSnowden33

nathanRamaNoodles, 2020-05-09:
Oh man, you fixed my one-week issue.  How on earth did you figure out that it was the size of the MTU?
Either way, thanks @JSnowden33 !