dekuNukem/Nintendo_Switch_Reverse_Engineering #84: Joycon joystick dimensions

honzapatCZ, 2019-08-15:
Iam currently working on a small project DIY controller and I would like to add the switches iconic small factor joystick so I ordered some from china, sadly it would take a lot of time(month) till they come and iam already working on the model, so can some great soul take me those measurements?
[image]
(preferably into the image)

wawa1474, 2019-12-01:
take these measurements with a grain of salt...
but they are for the most part close to correct.
[image]

Nori1313, 2021-02-15:
Used a 0.063 inch pin gauge and a 1.6 millimetre drill bit to obtain hole size from unofficial replacement joystick; hole size could be 1.65 mm, but 0.05 mm is negligible in this situation. Measured around 14.90 mm to 15.00 mm for the joystick diameter across five different samples (4 official, 1 unofficial), and confirmed all other measurements by wawa1474.
[image]

Kazumasan, 2022-05-14:
I've compiled it to this drawing i think this is neater and easier to understand.

[image]

rinscr3003, 2024-07-21:
In my practical trial, for most AliExpress components this gap circled in blue should be 0.1mm, not as the above one's 0.3mm
[image]