dekuNukem/Nintendo_Switch_Reverse_Engineering #87: Right joycon pinout?

SamanthaRenamon, 2019-10-10:
Do you have the schematics of the right joycon pinout i need to make a bridge for the r button

ratijas, 2019-10-30:
Yeah, need to fix the R button myself. R button box teared off together with its three pins. Seems like the only choice is to remove the PBC lacquer and attempt to solder those extra-thin cooper threads. Unless there are alternative pins somewhere...

SamanthaRenamon, 2019-11-01:
> Yeah, need to fix the R button myself. R button box teared off together with its three pins. Seems like the only choice is to remove the PBC lacquer and attempt to solder those extra-thin cooper threads. Unless there are alternative pins somewhere...
> [image]
> [image]

I acctually fix it

ratijas, 2019-11-05:
@SamanthaRenamon Could you be a little bit more specific, please? I still haven't fixed mine, but figured out a lot in the process of trying.

jul666, 2019-11-30:
Hey !
I've break the same button.
There is an alternate pinout possibility
[image]

numselli, 2022-03-05:
I have this same issue how did you fix it[image]

jul666, 2022-03-06:
[image]
i've fixed mine by using ratijas solution.
- removing lacquer on the track
- soldering the button in the good position with left and right pad
- soldering tiny cooper wire on the track to the button

numselli, 2022-03-06:
Did you use hot air or an iron?

numselli, 2022-03-07:
Also is this how the traces connect?[image]