dekuNukem/Nintendo_Switch_Reverse_Engineering #97: Output from ring-con?

maffe03, 2020-04-10:
Are there any plans to find out how to read the output from the ring-con when it's attached? I'm guessing it's an axis, but I don't even know where to start looking. Does anyone know?

Yamakaky, 2021-02-08:
I made some progress following https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/98. After configuration, the ringcon flex is sent instead of the accelerometer in the third IMU packet in the standard input report. The gyro and 2/3 of the accelerometersstays the same. I'm looking to replicate it in https://github.com/Yamakaky/joy, will post when it works.

Yamakaky, 2021-02-12:
It works, you can run `cargo run --bin joytk ringcon` with a paired joycon. It shows the raw hex value from the ringcon. I can also read the number of offline flex (when you press the stick with the switch off), there is what seem to be a crc but I can't find how it works. I can only write a flex count if I already know the CRC. Here are some values I measured:
https://0bin.net/paste/t6RlxsgX#YR4iCrD-sO/UB1RWx2VY9cPBmJUP/Ira1S6QtrRpmg0. At 0x1f3/500, the ringcon auto turns off.

Yamakaky, 2021-02-12:
The setup process is as follow: https://github.com/Yamakaky/joy/blob/master/joycon/src/hid.rs#L443. I don't know what most of these commands do XD.

Kurainu, 2021-02-12:
Oh nice good to hear you got progress. I also wanted to try to make sense how to get the flex sensor data etc. 
maybe i find some time this weekend to play around with the ringcon myself and hopefully get figure out something useful 😄.

Yamakaky, 2021-02-12:
I recorded some traces, you can use my tool `cargo run --bin joytk decode < trace/x.log`. It filters out the rumbleonly output and full input reports, basically only subcommands.

Kurainu, 2021-02-12:
I think i use your Tool and i try to get the relay/proxy thingy up and running that i can get a look at the packets in realtime or do my own trace.

Yamakaky, 2021-02-12:
I have to do a readme XD

Kurainu, 2021-02-12:
Yeah do it  or i try and error my way Through it xD (i play CTFs i have endurance on must :D). or i try to use the python script from the repo in the #98

Yamakaky, 2021-02-12:
You have to use the python script for capture, my tool only does the decoding part using my parsing lib.

Yamakaky, 2021-02-13:
I added some doc for the cli and libraries.

Yamakaky, 2021-02-14:
And I added a subcommand to proxy and record, all in rust! It's pretty raw so testing would be appreciated ;)

Kurainu, 2021-02-15:
I tried the python script but i dont get the relay to send the packets from the Joycon/Ringcon through my Linux Laptop to my Switch. Its mostly Stuck at the SDP Adverstising but if it goes past it co conection to my Switch is made. Is it intended to work or i'm missing something? .
But i can Capture the packets from the Joycon to a log file and use your Rust Script to decode them. And i Test the proxy and record Thingy when i get home

The rest of my Remaining free Weekend i tried first to understand what your rust script sends to activate the ringcon but its a slow progress because i dont know rust that well. Because when i have those i can tinker with it in a language that i know better than rust :D. If you could send me those commands as bytes if would be Great and saves my quiet some time getting those @Yamakaky :)

Yamakaky, 2021-02-15:
Sure, I wanted to add logging anyway XD

I did my joycon lib will structs everywhere which makes it very convenient to use, but can make it hard to figure out the exact bytes sent.

Yamakaky, 2021-02-15:
I added full logging support, `env RUST_LOG=debug cargo run --bin joytk ringcon monitor` now show the raw reports \o/

ringrunnermg, 2021-02-19:
I have just made public a working implementation of the Ringcon to PC software - https://github.com/ringrunnermg/Ringcon-Driver. Any feedback would be great. 

Also, good work Yamakaky, it took me months to get the data and it wasn't as complete as what you have got.

Yamakaky, 2021-02-19:
oooh, you found the crc table for the ringcon? nice! Out of curiosity, how did you find it?

And yeah, I see everyone using raw buffer manipulation and copy pasted 20 lines loops while I'm like https://github.com/Yamakaky/joy/blob/master/joycon/src/hid.rs#L477 ^^

BTW you can detect if the ringcon is plugged by reading the Connection Info field in the standard input report. See https://github.com/Yamakaky/joy/blob/master/joycon-sys/src/input.rs#L225 for the additional values. It seems there is only one accessory that plugs to the ringcon so a lot of it is guess work. See also https://github.com/Yamakaky/joy/blob/master/joycon-sys/src/accessory.rs for what I could find about the protocol.

I still have to do some testing on which subcommands are required. I have a tight schedule for now so it may take some time ^^

WitoTV, 2021-02-21:
Do you guys think we could get this as a "raspberry pi zero w" USB-connected device?
Aka. presents itself as keyboard + mouse or controller?
Of course, some kind of desktop app would be needed to make a setup, select output type, bind actions, etc...

I kind of want to make that but all the buffer shenanigans are too much my front-end developer brain.
(I may do it but it may take forever to fully understand the whole buffer stuff)

aka. joycons - (bluetooth) -> pi -> (usb) -> pc/console.

Yamakaky, 2021-02-21:
Actually, that seem very doable:
- I already have a WIP gamepad to mouse/keyboard mapper, you just need to update https://github.com/Yamakaky/joy/blob/master/gyromouse/src/main.rs#L113-L131 for the key presses. It also supports the dualshock, and https://docs.rs/gilrs could be used to add other gamepads. It doesn't support the ringcon which is very joycon specific, so that would need to be added. 
- The virtual keyboard part seems easy, for example https://randomnerdtutorials.com/raspberry-pi-zero-usb-keyboard-hid/
I can guide you through it at https://github.com/Yamakaky/joy/discussions.

hoverbird, 2021-03-13:
Amazing work @Yamakaky and @ringrunnermg, I've been trying to get at this data (off and on, I'm no hardware hacker) for over a year with very little success... so excited y'all have really cracked this! 👏👏👏👏👏👏

xinzhuwei, 2021-11-05:
@Yamakaky  I feel that gyroscope and timestamp are fused，

Yamakaky, 2021-12-05:
What do you mean by that?

xinzhuwei, 2021-12-06:
the RING CON is enabled  ，joy con Gyro data protocol？

xinzhuwei, 2021-12-15:
Enable IMU (6-Axis sensor)  JOY CON FW 4.17     subcmd  40 03     Gyro  ?

xinzhuwei, 2021-12-15:
@Yamakaky

Yamakaky, 2021-12-15:
I'm sorry @xinzhuwei but I'm not sure what you mean by that? You indeed have to call subcommand 40 with parameter 03, which I called MaybeRingcon here https://github.com/Yamakaky/joy/blob/2691f8c655213672eb385adaae5a073353a0177c/crates/joycon/src/hid.rs#L487.

xinzhuwei, 2021-12-20:
@Yamakaky  Yes, in mayberingcon mode, the data and timestamp of the gyroscope change. I don't know the change protocol. It seems to be related to subcommand 5C

xinzhuwei, 2022-01-07:
@Yamakaky 
Yes, in MaybeRingcon  ；
In the Bluetooth data report, is the gyroscope data related to the time stamp?
but ,I don’t know the protocol of gyroscope data and time stamp

a1306440000000ec877bde67790a8dfd3e0212101700f6ff0c006dfd7e010f101200fcfff0ff55fd5900191011000000e4ff

german77, 2022-12-04:
Anyone has been able to pull calibration and step data from the ring?

Meorge, 2025-05-03:
Has anyone been able to do a write-up of how connecting to the Ring-Con works? I see people have published code to do it, in joy and Ringcon-Driver, but so far this thread appears to be the only place where people have talked about the steps for accomplishing it, and it seems like the info here is incomplete 😅

german77, 2025-05-04:
Sure.

Prepare the joycon MCU:
- Set report mode to Full 60Hz(0x30)
- Enable MCU
- Configure the MCU to STANDBY 

Actually enable the device:
- Optional step: Check if the ring is connected by sending the GET_EXTERNAL_DEVICE_INFO(0x59) sub command. Ring id is 0x2000
- Send SET_EXTERNAL_FORMAT_CONFIG(0x5c) subcommand with this data. It might be calibration data I'm not sure.
```
0x06, 0x03, 0x25, 0x06, 0x00, 0x00, 0x00, 0x00, 0x1C, 0x16, 0xED, 0x34, 0x36,
0x00, 0x00, 0x00, 0x0A, 0x64, 0x0B, 0xE6, 0xA9, 0x22, 0x00, 0x00, 0x04, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x90, 0xA8, 0xE1, 0x34, 0x36
```
- Send ENABLE_EXTERNAL_POLLING(0x5A) with this data `0x04, 0x01, 0x01, 0x02`. This enables polling from the sensor data until the MCU is disabled

german77, 2025-05-04:
I was able to capture some packets to send actual commands to the ring and joycon(see image). To enable pull/press/step counters while on sleep mode. However I never wrote any code for it as the emulator is dead. Feel free to ask if you want to continue development of this device.

[image]

Meorge, 2025-05-04:
Thank you very much for this info! 😄 It looks like I'll need to get the MCU configuration and CRC stuff figured out, which might take a bit. For now my goal is just to get the current flex of the Ring-Con (not worried about the stored presses at the moment).

german77, 2025-05-04:
You can find the crc table here https://github.com/CTCaer/jc_toolkit/blob/master/jctool/ir_sensor.h most likely how to calculate it as well

Meorge, 2025-05-04:
After adding the CRC table bit, it's now working! Thank you very much!! 🥳 🎉

HelloRayH, 2025-06-11:
> Mode 1 is enough to get things working. But in case you are interested mode 2 format
> 
> ```
> typedef struct{
>     int16_t y;
>     int16_t x;
>     int16_t z;
> } vector_s;
> 
> typedef struct {
>     vector_s accel_0;
>     uint32_t mode : 2;
>     uint32_t max_index : 2;
>     uint32_t last_sample_0 : 21;
>     uint32_t last_sample_1l : 7;
>     uint16_t last_sample_1h : 14;
>     uint16_t last_sample_2l : 2;
>     vector_s accel_1;
>     uint32_t last_sample_2h : 19;
>     uint32_t delta_last_first_0 : 13;
>     uint16_t delta_last_first_1 : 13;
>     uint16_t delta_last_first_2l : 3;
>     vector_s accel_2;
>     uint32_t delta_last_first_2h : 10;
>     uint32_t delta_mid_avg_0 : 7;
>     uint32_t delta_mid_avg_1 : 7;
>     uint32_t delta_mid_avg_2 : 7;
>     uint32_t timestamp_start_l : 1;
>     uint16_t timestamp_start_h : 10;
>     uint16_t timestamp_count : 6;
> } mode_2_s;
> ```

@german77 

Thank you very much for your exciting discovery.

I don't know if my understanding is correct, please help me find the cause of the problem.
According to the structure definition of mode 2, according to my understanding, under mode 2, the data obtained during the calibration process, after decoding the structure of mode 2, should be some values ​​that change around the offset.

The following data is part of the data log generated by the joycon imu during the calibration process.
0xCD,0xFF,0xFC,0xFF,0x42,0xF0,0xE2,0x42,0x92,0xE1,0x8A,0x7C,0xCA,0xFF,0xFA,0xFF,0x3D,0xF0,0xAB,0x74,0xAE,0x00,0xE1,0x3F,0xC8,0xFF,0xFE,0xFF,0x41,0xF0,0x01,0x04,0x02,0x80,0x56,0x10
0xCE,0xFF,0xFA,0xFF,0x3F,0xF0,0x22,0x45,0x92,0x6B,0x8A,0x3C,0xCD,0xFF,0xFC,0xFF,0x42,0xF0,0xB0,0x74,0x96,0x00,0xE3,0x5F,0xCA,0xFF,0xFA,0xFF,0x3D,0xF0,0x01,0x04,0x02,0x80,0x58,0x10
0xD1,0xFF,0xF6,0xFF,0x3F,0xF0,0x82,0x47,0x92,0xF5,0x89,0xFC,0xCE,0xFF,0xFA,0xFF,0x3F,0xF0,0xB4,0x74,0x9E,0x00,0xE3,0x5F,0xCD,0xFF,0xFC,0xFF,0x42,0xF0,0x01,0x04,0x02,0x81,0x5A,0x10
0xD1,0xFF,0xF3,0xFF,0x40,0xF0,0xE2,0x49,0x92,0x79,0x89,0x7C,0xD1,0xFF,0xF6,0xFF,0x3F,0xF0,0xB9,0x74,0xA6,0x00,0xDF,0x3F,0xCE,0xFF,0xFA,0xFF,0x3F,0xF0,0x01,0x04,0x00,0x80,0x5C,0x10
0xCD,0xFF,0xF7,0xFF,0x3E,0xF0,0x62,0x4C,0x92,0xFB,0x88,0x7C,0xD1,0xFF,0xF3,0xFF,0x40,0xF0,0xBE,0x74,0x9E,0x00,0xE1,0x3F,0xD1,0xFF,0xF6,0xFF,0x3F,0xF0,0x01,0x04,0x02,0x80,0x5E,0x10
0xCA,0xFF,0xF8,0xFF,0x3D,0xF0,0xB2,0x4E,0x92,0x85,0x88,0x7C,0xCD,0xFF,0xF7,0xFF,0x3E,0xF0,0xC3,0x74,0x9E,0x00,0xE2,0x7F,0xD1,0xFF,0xF3,0xFF,0x40,0xF0,0x01,0x04,0x02,0x80,0x60,0x10
0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0x22,0x51,0x92,0x0F,0x88,0x3C,0xCA,0xFF,0xF8,0xFF,0x3D,0xF0,0xC8,0x74,0x9E,0x00,0xE4,0x3F,0xCD,0xFF,0xF7,0xFF,0x3E,0xF0,0x01,0x04,0x02,0x81,0x62,0x10
0xCB,0xFF,0xFA,0xFF,0x3C,0xF0,0x82,0x53,0x92,0xA5,0x87,0xBC,0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0xCD,0x74,0x96,0x00,0xE4,0x7F,0xCA,0xFF,0xF8,0xFF,0x3D,0xF0,0x01,0x04,0x02,0x81,0x64,0x10
0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0x02,0x56,0x92,0x29,0x87,0x3C,0xCB,0xFF,0xFA,0xFF,0x3C,0xF0,0xD3,0x74,0xA6,0x00,0xE2,0x7F,0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0x01,0x04,0x02,0x80,0x66,0x10
0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0x52,0x58,0x92,0xB3,0x86,0xFC,0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0xD8,0x74,0x9E,0x00,0xE2,0x7F,0xCB,0xFF,0xFA,0xFF,0x3C,0xF0,0x01,0x04,0x02,0x80,0x68,0x10
0xCA,0xFF,0xFA,0xFF,0x3D,0xF0,0xB2,0x5A,0x92,0x35,0x86,0xFC,0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0xDD,0x74,0x9E,0x00,0xE1,0x5F,0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0x01,0x00,0x02,0x81,0x6A,0x10
0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0x32,0x5D,0x92,0xC3,0x85,0xBC,0xCA,0xFF,0xFA,0xFF,0x3D,0xF0,0xE2,0x74,0xA6,0x00,0xE3,0x5F,0xC9,0xFF,0xF9,0xFF,0x3D,0xF0,0x01,0x04,0x00,0x81,0x6C,0x10
0xC8,0xFF,0xF9,0xFF,0x40,0xF0,0xC2,0x5F,0x92,0x43,0x85,0x7C,0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0xE7,0x74,0x9E,0x00,0xE0,0x3F,0xCA,0xFF,0xFA,0xFF,0x3D,0xF0,0x01,0x00,0x00,0x81,0x6E,0x10
0xC9,0xFF,0xFB,0xFF,0x3D,0xF0,0x42,0x62,0x92,0xC3,0x84,0xBC,0xC8,0xFF,0xF9,0xFF,0x40,0xF0,0xEC,0x74,0xA6,0x00,0xE0,0x5F,0xCC,0xFF,0xFA,0xFF,0x3E,0xF0,0x01,0x04,0x02,0x80,0x70,0x10
0xC9,0xFF,0xFA,0xFF,0x3E,0xF0,0xC2,0x64,0x92,0x4B,0x84,0xBC,0xC9,0xFF,0xFB,0xFF,0x3D,0xF0,0xF0,0x74,0xA6,0x00,0xE2,0x3F,0xC8,0xFF,0xF9,0xFF,0x40,0xF0,0x01,0x04,0x02,0x80,0x72,0x10
0xC9,0xFF,0xF9,0xFF,0x3E,0xF0,0x02,0x67,0x92,0xCF,0x83,0x7C,0xC9,0xFF,0xFA,0xFF,0x3E,0xF0,0xF5,0x74,0x8E,0x00,0xE1,0x7F,0xC9,0xFF,0xFB,0xFF,0x3D,0xF0,0x01,0x04,0x02,0x80,0x74,0x10
0xC8,0xFF,0xF9,0xFF,0x3F,0xF0,0x72,0x69,0x92,0x5B,0x83,0xFC,0xC9,0xFF,0xF9,0xFF,0x3E,0xF0,0xF9,0x74,0xA6,0x00,0xE3,0x1F,0xC9,0xFF,0xFA,0xFF,0x3E,0xF0,0x01,0x04,0x02,0x81,0x76,0x10
0xCA,0xFF,0xFC,0xFF,0x40,0xF0,0xF2,0x6B,0x92,0xE3,0x82,0x7C,0xC8,0xFF,0xF9,0xFF,0x3F,0xF0,0xFE,0x74,0xA6,0x00,0xE2,0x1F,0xC9,0xFF,0xF9,0xFF,0x3E,0xF0,0x01,0x04,0x02,0x81,0x78,0x10
0xC9,0xFF,0xF9,0xFF,0x3F,0xF0,0x82,0x6E,0x92,0x65,0x82,0x3C,0xCA,0xFF,0xFC,0xFF,0x40,0xF0,0x02,0x75,0xAE,0x00,0xE1,0xFF,0xC8,0xFF,0xF9,0xFF,0x3F,0xF0,0x00,0x00,0x02,0x80,0x7A,0x10
0xCA,0xFF,0xF8,0xFF,0x3F,0xF0,0xE2,0x70,0x92,0xE7,0x81,0xBC,0xC9,0xFF,0xF9,0xFF,0x3F,0xF0,0x06,0x75,0x9E,0x00,0xE0,0x5F,0xCA,0xFF,0xFC,0xFF,0x40,0xF0,0x01,0x04,0x00,0x80,0x7C,0x10
0xCB,0xFF,0xF9,0xFF,0x3D,0xF0,0x52,0x73,0x92,0x6D,0x81,0xBC,0xCA,0xFF,0xF8,0xFF,0x3F,0xF0,0x0B,0x75,0x9E,0x00,0xE1,0x3F,0xC9,0xFF,0xF9,0xFF,0x3F,0xF0,0x01,0x04,0x02,0x81,0x7E,0x10
0xC8,0xFF,0xF9,0xFF,0x39,0xF0,0xC2,0x75,0x92,0xF5,0x80,0x3C,0xCB,0xFF,0xF9,0xFF,0x3D,0xF0,0x11,0x75,0xA6,0x00,0xE1,0x5F,0xCA,0xFF,0xF8,0xFF,0x3F,0xF0,0x01,0x00,0x02,0x81,0x80,0x10
[... cut ...]
0xCA,0xFF,0xF9,0xFF,0x49,0xF0,0x22,0x89,0x92,0x2F,0x7D,0x7C,0xC9,0xFF,0xF9,0xFF,0x43,0xF0,0x36,0x75,0xA6,0x00,0xE4,0x3F,0xC8,0xFF,0xF7,0xFF,0x3C,0xF0,0x01,0x04,0x02,0x81,0x90,0x10
0xCA,0xFF,0xF9,0xFF,0x43,0xF0,0xB2,0x8B,0x92,0xB9,0x7C,0x3C,0xCA,0xFF,0xF9,0xFF,0x49,0xF0,0x3B,0x75,0xA6,0x00,0xE2,0x5F,0xC9,0xFF,0xF9,0xFF,0x43,0xF0,0x01,0x04,0x02,0x81,0x92,0x10
0xCA,0xFF,0xF9,0xFF,0x3A,0xF0,0x22,0x8E,0x92,0x45,0x7C,0xFC,0xCA,0xFF,0xF9,0xFF,0x43,0xF0,0x3F,0x75,0xA6,0x00,0xE3,0x3F,0xCA,0xFF,0xF9,0xFF,0x49,0xF0,0x01,0x04,0x00,0x80,0x94,0x10

After mode2 decoding, the generated gyro_x, gyro_y, and gyro_z graphs are as follows. Did I get something wrong?

[image]

HelloRayH, 2025-06-12:
I also found that since the controller firmware was updated to 4.00 or above, on the SWITCH game console, 40 02/03 is sent to request imu data.
01 0C 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
21 36 91 00 80 00 94 A8 73 49 68 77 00 82 02 04 21 03 02 E0 F6 B5 0A D7 63 03 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

01 0A 00 00 00 00 00 00 00 00 40 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
21 9E 91 00 80 00 93 B8 73 49 68 77 0A 80 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

So I used jc_toolkit from @CTCaer , modified the parameters of subcommand 40 to 02/03, and obtained imu data in mode 02/03. The only difference between the two modes is that in mode 03, the last group of accel data is filled with six 00s. I think this is the named MaybeRingcon defined by @Yamakaky .

[image]

HelloRayH, 2025-06-13:
> Mode 3 is a quaternion if I recall correctly. Unless If I mixing things. I personally just use mode 1 regardless of the use case. But give me a few days and I will try to parse the data myself and see if I get reasonable results as your graphs do look quite jittery.

Yeah, that's what confuses me.

HelloRayH, 2025-06-19:
@german77 
  Hello.
      I record the data of the pro controller when it is still and swinging yaw (place on the flat table. Obviously, although it is mainly the change of yaw, it will inevitably affect pitch and roll), and parse each member of the structure according to mode_2_s, and get the following graph. 
    When the Pro controller is still, these members (last_sample_0, last_sample_1, last_sample_2) seem more like a counter or timestamp, but when the handle moves, they will change accordingly.I have no idea how to get the real data (roll, pitch, yaw) from these logs yet. 

[image]

[image]

[image]

[image]

[image]

[image]