dekuNukem/Nintendo_Switch_Reverse_Engineering #99: Poke Ball Plus

Davidobot, 2020-04-15:
Anyone got one of these babies and tried to see if the protocol it uses is similar to the joy/procon one?

Here's a related issue on my program: https://github.com/Davidobot/BetterJoy/issues/299

Looks like there is gyro/accel data, and also sound-in! I don't have the controller, but it would be neat if someone could investigate.

VictorGamer072YT, 2020-05-14:
If you still need help, I have a Pokeball Plus and I'm willing to help!

Sritterbush, 2020-06-30:
I have one I don't mind disassembling for the cause, but I'll need someone on video to oversee the process and ensure I gather the right data.

Yamakaky, 2021-02-08:
@Sritterbush do you still have it? I'm [making progress](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/98), so if you could run the capture I could analyze it.

eproasim, 2023-01-31:
Hi guys! I came across this issue while trying to read the Poke Ball Plus values in Unity. I'm not quite ready to say it's solved completely, but I think it's close. Check it my findings here: https://www.reddit.com/r/Unity3D/comments/10pjw91/update_poke_ball_plus_in_unity/