dekuNukem/Nintendo_Switch_Reverse_Engineering #102: IMU unit noise floor

Yamakaky, 2020-05-18:
I'm trying to understand what https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/imu_sensor_notes.md#noise-level-range-for-each-sensitivity-configuration means. Does it mean that each measurement is in the range +- this value, and that 0 +- this value can be considered 0? If your values are true, that means the noise level grows linearly with the precision, so why use other sensitivity settings?

Also, where does these values come from? I couldn't find them from the datasheet.

Finally, where did you see the explanation about the need to add 15% for the gyro? The datasheet only mentions the values with 15% applied. It seems it would only give angles that are 15% off.

xinzhuwei, 2022-05-04:
@Yamakaky When joy con firmware version <4.86 ：Subcommand 0x40: Enable IMU (6-Axis sensor)
One argument of x00 Disable or x01 Enable.  When joy con firmware version ==4.17 ：Subcommand 0x40: Enable IMU (6-Axis sensor)
One argument of x00 Disable , x01 mode1. 0x02 mode2  0x03   mode3      Do you know the protocol of mode 2 and mode 3？

Yamakaky, 2022-05-05:
No, sorry...