dekuNukem/Nintendo_Switch_Reverse_Engineering #106: Original vs v2

Sritterbush, 2020-06-30:
A comparison of the v2 Joycons to the originals. While I expect they'll mostly be the same, some minor hardware changes have been observed and knowing whether there are other differences would be highly valuable to the community. One theory is that the v2's hardware tweaks reduced drift through placement of raised surfaces, but if there are other edits with that they have been missed.

Sritterbush, 2020-06-30:
To clarify, you can identify the newer Joycons by their model number.
Original Joycon - HAC-A-JCL-C1 (JoyCon Left - Configuration 1)
Padded Joycon - HAC-A-JCL-C2 (JoyCon Left - Configuation 2) [No hardware changes, added some padding]
v2 Joycon - HAC-A-JCR-C3 (JoyCon Right - Configuration 3) [Yes, technically 3rd generation but nobody calls it v3]

AngheloAlf, 2020-10-13:
Does a "configuration 0" exists? 

I have a pair of joycons that are labeled HAC-A-JCL-C0 (and JCR-C0 for the right one). Or am I not looking to the model number?

AngheloAlf, 2020-10-13:
[image]