dekuNukem/Nintendo_Switch_Reverse_Engineering #124: Can the Joycon joystick work with 5v?

HellSingCoder, 2021-07-16:
I am aware of the standard pinout with the 3.3v input but, would it be possible to make it work with 5v instead? will it cause any damage to the internal components of the joystick itself?

superpotato9, 2021-07-16:
If you simply want to run 5v through it then you should be fine

EuphoricPenguin, 2021-10-11:
Is there a guide anywhere for this? I'm struggling to find information on the acceptable voltage/current range.

HellSingCoder, 2021-11-22:
I can confirm it works with 5V. I have been running it for about a month now, multiple hours a day.

EuphoricPenguin, 2021-11-24:
> I can confirm it works with 5V. I have been running it for about a month now, multiple hours a day.

That's nice to hear. Was there ever a document to suggest otherwise? I've combed over the issue threads here several times, and I've never found the pinout (with 3.3v) you were talking about. I found an image where someone had added 3.3V to the silkscreen, but that was it.

Aezore, 2021-11-24:
I haven’t done any work on this but I do work with electronics and while
it’s advised to stick to specs it’s possible to overdrive inputs as devices
tend to be protected as long as you don’t go crazy with both current and
voltages.
One of the main reasons portable electronics do work with 3.3v nowadays is
because lithium batteries are fully charged at 3.5/3.7v and that gives room
to operate safely without extra regulation circuitry. Also temperatures and
trace widths and all sort of noise stuff.

If you’re worried you could try to build a voltage clamp with a pair of
schotky diodes to the supply rail. It’s basically two diodes together in
parallel with the rail and ground and the diodes junction being the node
for the signal wire. That sinks any voltage above the supply rail and
protects the device. It’s usually a protection measure instead of a
“feature” but should do the job. Also there are cheap level shifters in the
market for that matter.