dekuNukem/Nintendo_Switch_Reverse_Engineering #126: Joycon power circuit

klimaszewskitom, 2021-07-19:
I think I short-circuited battery connector on my right joycon an it doesn't power up now. Battery is charged (multimeter outputs 4V). When battery is attached the only component I see it gets power is a resistor right to the battery connector.
[image]
It also gives ~4V.

 I didn't done full diagnosis yet to check if anything is shorted. Do you know about any fuse on the PCB or have idea about it's power-on process? Or maybe somebody had similar issue and have any ideas what can I do? I have a really good soldering skills so it won't be a problem for me to replace any capacitor or resistor there.

bigla, 2021-12-18:
hey @klimaszewskitom could you resolve the problem? I have the exact same one.

Did you compare it with an original working one?

It's so sad there are no schematics or service manuals to find anywhere...

klimaszewskitom, 2021-12-20:
Hi @bigla, no, unfortunately I haven't. I still have this joycon so if you would like to cross check something with me I'm happy to cooperate :)

bigla, 2021-12-21:
I am taking measurements at every testpoint I see on this side (of a known working one) and will comment with a photo of the measurements.

Sadly it won't look beautiful or professional, but maybe it will help with diagnosis.

I already found some differences between my broken one and the good one, resolving them will be the more intesting part.

By the way, do you have an idea how i should measure testpoints which don't have a fixed (DC) voltage? (I assume they can either be open loop - reference is bat-minus - or maybe some signal...) The value range is similar to measuring "air" being the cause of my assumption....

bigla, 2021-12-21:
By the way, make sure the fuse, the cap and the testpoint above the fuse are connected to bat+ (continuity)!

[image]

klimaszewskitom, 2021-12-27:
@bigla  glad that you've made it :) I will try with mine as soon as I get new multimeter, my old one broke.

> By the way, do you have an idea how i should measure testpoints which don't have a fixed (DC) voltage? (I assume they can either be open loop - reference is bat-minus - or maybe some signal...) The value range is similar to measuring "air" being the cause of my assumption....
You need a logical state analyzer for that. It should be able to register all the voltage changes over time.

bigla, 2022-01-14:
@klimaszewskitom Well I do have the tools and I am eager to learn. If you want to take the time to tutor me through the process I can do it for you (and anyone reading this).