dekuNukem/Nintendo_Switch_Reverse_Engineering #130: Joycon charging grip question

ssteejans, 2021-08-19:
I am working on a handheld raspberry pi based retro system and am going to use joycons as my input device. Based on the info in the USB-HID-Notes, I got a charging grip to take apart and use for integrating the joycons. However, it seems like the grip only charges the joycons and does not communicate via usb.

Does the charging grip allow for the use of the joycons over usb? 

One further detail is that the charging grip pcb (EGB-01 2017) has almost nothing populated on it. The image of the pcb that @shinyquagsire23 posted [here](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/7#issuecomment-292707083) shows a fully populated pcb with a different layout. [This](https://www.amazon.com/gp/product/B0967N2GZ5/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) is the grip I purchased.
[image] 

[image]

ssteejans, 2021-09-17:
I have some further information on this.
The charging grip shown above was sold by "Nintendo NSW" on amazon and was bulk packaging. It does not allow for joycons over usb.

I purchased another charging grip from gamestop in the USA with standard packaging which contains the pcb that shinyquagsire23 posted images of.  This one DOES work over usb.

OfficialEduardoLuna, 2022-11-26:
> I have some further information on this. The charging grip shown above was sold by "Nintendo NSW" on amazon and was bulk packaging. It does not allow for joycons over usb.
>
> I purchased another [...]
@ssteejans  did it end up working for you? do you have pictures of the final build?

I too want to build something like this. I wanted to try to use hori split pad though so I may use their attachment instead

Tokobotenkai, 2023-11-28:
> I have some further information on this. The charging grip shown above was sold by "Nintendo NSW" on amazon and was bulk packaging. It does not allow for joycons over usb.
>
> I purchased another [...]

Not sure if this is the right place to ask, but may I know does Steam recognise the joycons as a controller when you connect the charging grip to a PC?