dekuNukem/Nintendo_Switch_Reverse_Engineering #131: How to re-connect a Joycon?

ItachiSan, 2021-09-14:
Hi all,

I am currently using [BetterJoy](https://github.com/Davidobot/BetterJoy) and I am quite happy about it, although I do have currently a problem; when turning off a controller by long-pressing the home/capture button for 2 seconds or the sync button for 1, the device forgets about the Bluetooth pairing and requires to be re-paired.

I've been digging in the code and I am fairly confident that I can adjust the HID commands for setting up pairing, but that leads me to be unsure about the pairing process.
I read in [the section about manual pairing](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_subcommands_notes.md#subcommand-0x01-bluetooth-manual-pairing) but I am a bit unsure about the subcommands after the first twos.

To summarize:
1. Is it possible to keep the connection between a PC and a (pair of) Joycon(s)?
2. If (1), are the steps in the guide above still valid? What is the expected response between the two devices after the first 2 subcommads?
3. Is there any special (sub)command to keep an eye on when the Joycon turns up again?
4. If pressing the sync button resets the pairing, is it possible to put the Joycon in a sleep state? If so, how?

Tschuuuls, 2022-10-25:
I'm currently looking into the linux Kernel Driver hid_nintendo.
Reconnecting is also fairly broken there, at least with multiple controllers connected it will often fail.
What works is pressing a button to wake the controller, it will then cycle through the leds. Turn the controller off by short pressing the sync button. Press a random button and it will be able to reconnect.
Seems like the controller is in a weird state when disconnected from the PC and needs to be rebooted somehow.
The question is now: How to prevent entering this state.

Userlinger, 2022-10-27:
When the controller falls "asleep" i can wake it up with a buttonpress and it reconnects. But after a reboot of the pc that does not work anymore. that leads me to believe that the joycon causes the bluetooth stack of the os things to be stored in a kind of cache that is lost on reboot. 

sometimes the connection status switches to connected briefly even after reboot, but a working connection is never established.

nahlahman, 2023-04-29:
In my experience and that of others, Bluetooth 5.X adapters have this issue with Joy-cons, while Bluetooth 4.X adapters do not. 

You can take it with a grain of salt, because the cause could be something else. But for me two 5.X adapters had this issue and my 4.X adapter does not. I've seen similar comments across the web. I've also seen a few comments contradicting it, but without elaboration.

Userlinger, 2023-05-05:
could you tell the model of a blueooth 4.x adapter that worked for you?

yukihiroK, 2023-12-25:
I encountered a similar phenomenon with [JoyconLib](https://github.com/Looking-Glass/JoyconLib), not BetterJoy, and found a solution.

The phenomenon I encountered was that after running the program in Play Mode on the Unity editor, connecting it to the Joy-Con, and then finishing the execution, once the Joy-Con went to sleep, I could not reconnect it again.
This was clearly a problem with the program running in Unity, because if JoyCon went to sleep before running Unity, I was able to reconnect it.

As a result, it turns out that there is a problem with the way JoyconLib [Detach](https://github.com/Looking-Glass/JoyconLib/blob/cf8850deef20867d0f51e0228b9298b853ca4cff/Packages/com.lookingglass.joyconlib/JoyconLib_scripts/Joycon.cs#L274) the JoyCon.

Sending a [Reconnect command](https://github.com/dekuNukem/Nintendo_Switch_Reverse_Engineering/blob/master/bluetooth_hid_subcommands_notes.md#subcommand-0x06-set-hci-state-disconnectpagepairturn-off) to JoyCon when exiting an application will automatically disconnect the JoyCon and then reconnect it properly.

```.cs
public void Detach()
{
    stop_polling = true;
    PrintArray(max, format: "Max {0:S}", d: DebugType.IMU);
    PrintArray(sum, format: "Sum {0:S}", d: DebugType.IMU);
    if (state > state_.NO_JOYCONS)
    {
        Subcommand(0x30, new byte[] { 0x1 }, 1);
        Subcommand(0x40, new byte[] { 0x0 }, 1);
        Subcommand(0x48, new byte[] { 0x0 }, 1);
        Subcommand(0x3, new byte[] { 0x3f }, 1);
        Subcommand(0x6, new byte[] { 0x1 }, 1); // Send Reconnect Command!
    }
    if (state > state_.DROPPED)
    {
        HIDapi.hid_close(handle);
    }
    state = state_.NOT_ATTACHED;
}
```