dekuNukem/Nintendo_Switch_Reverse_Engineering #134: Looking for ideas on joy con charging

mehtmehtsen, 2022-08-08:
I want to add charging functionality to a custom joy con holder I built. Ideally I’d be able to use the joy cons while charging, so harvesting the innards of the official charging joy con grip was the first idea I had. But as the joy cons will be mounted about 20cm apart and I couldn’t find any feasable way to extend the FPCs connecting the PCB to the rail contacts, I’m not sure how to continue.

Is there really no practical way to extend the FPCs length?

Instead, could I just solder 5V and GND of an USB C cable (or USB C connector breakout board) to the respective pins of the rail contacts and be fine (thanks to the charging circuits being integrated into the joy cons themselves)? 

Are there other charging grips or docks that could be hacked more easily?

TsilaAllaoui, 2023-01-12:
> I want to add charging functionality to a custom joy con holder I built. Ideally I’d be able to use the joy cons while charging, so harvesting the innards of the official charging joy con grip was [...]

I also want to know if i can solder wires to 5V and GND and add a micro usb or usb c port in the bottom part of the joycons for the charging. And does any charger work or does they have range for the amps? I really searched everywhere but no one did standalone intergated charging port for each joycons befire

CoinOperatedGadget, 2023-05-26:
> > I want to add charging functionality to a custom joy con holder I built. Ideally I’d be able to use the joy cons while charging, so harvesting the innards of the official charging joy con grip was the first idea I had. But as the joy cons will be mounted about 20cm apart and I couldn’t find any feasable way to extend the FPCs connecting the PCB to the rail contacts, I’m not sure how to continue.
> > Is there really no practical way to extend the FPCs length?
> > Instead, could I just solder 5V and GND of an USB C cable (or USB C connector breakout board) to the respective pins of the rail contacts and be fine (thanks to the charging circuits being integrated into the joy cons themselves)?
> > Are there other charging grips or docks that could be hacked more easily?
> 
> I also want to know if i can solder wires to 5V and GND and add a micro usb or usb c port in the bottom part of the joycons for the charging. And does any charger work or does they have range for the amps? I really searched everywhere but no one did standalone integrated charging port for each joycons before.

Not exactly sure if this is what you're looking for, but I recently designed a pcb that the FPC cables from a pair of switch rails connect to, and hooking up 5v and ground seems to charge the joycons fine while I use them for my other purposes.  I haven't done any testing for amperage or anything like that though.