dekuNukem/Nintendo_Switch_Reverse_Engineering #137: compile a list of joycon drivers and modules from different languages?

ericries, 2022-09-09:
As I've been learning how to program with the joycons, I've relied on this repository (obviously) but also on the many, many modules out there that use this information in different ways to help programmers of different languages. By combining ideas from different languages, I've been able to solve a number of problems in my own language's driver.

Would it be useful to other people to compile a list of these drivers and their capabilities? I'd be glad to make a pull request to get things started

For the benefit of others who may come by this issue in the future, here are a few of the resources I've especially found useful:

libraries:
joycon-python: https://github.com/tocoteron/joycon-python
joycon-rs (rust): https://github.com/KaiseiYokoyama/joycon-rs
JoyconLib: https://github.com/Looking-Glass/JoyconLib
joy-con-webhid: https://github.com/sho-hey-hey/joy-con-webhid
JoyCon-Driver: https://github.com/fossephate/JoyCon-Driver
joycon (go): https://github.com/riking/joycon

jc tools:
jc_toolkit: https://github.com/ctcaer/jc_toolkit/
HID-Joy-Con-Whispering: https://github.com/shinyquagsire23/HID-Joy-Con-Whispering

Chromium (browser) implementation of the gamepad api for joycons:
https://chromium.googlesource.com/chromium/src/+/HEAD/device/gamepad/nintendo_controller.cc

nennneko5787, 2022-10-29:
JoyShockLibrary(C++): https://github.com/JibbSmart/JoyShockLibrary/
JoyShockLibrary For HSP3: https://github.com/nennneko5787/JoyShockLibrary-For-HSP3

ericries, 2022-10-29:
Super helpful, I hadn't seen either of those. Thanks!