dekuNukem/Nintendo_Switch_Reverse_Engineering #141: Pinout for the joycon rail flex cable specifically?

PADBX, 2022-12-04:
Hi,

So I know that the joycon rails on the switch have 10 pins to connect to the joycons but the flex cable that comes out of it has 11 pins. Is there a map of these pins anywhere? I would like to build some joycon extension cables.

PADBX, 2023-03-24:
Yeah I did, let me see if I can find my map.

Were you able to figure this out? I'm looking to make a specialized adapter that I need this pinout for.

CoinOperatedGadget, 2023-03-24:
Hope you're able to find it and much appreciated!

PADBX, 2023-03-24:
[image]
[image]
There that should be it. Thankfully the numbering is in order.

CoinOperatedGadget, 2023-05-26:
> [image] [image] There that should be it. Thankfully the numbering is in order. Turns out that one the pins has two heads.

Late but much obliged!  Finally got back to my own project and everything seems to work for my joy-con wired grip!

[image]

PointerOffset, 2023-08-04:
@CoinOperatedGadget Did you happen to use some kind of breakout board for working with the rail ribbon connectors? Would love to know a source for a connector if you have one.

CoinOperatedGadget, 2026-02-04:
> [@CoinOperatedGadget](https://github.com/CoinOperatedGadget) Did you happen to use some kind of breakout board for working with the rail ribbon connectors? Would love to know a source for a connector if you have one.

I know this is from forever ago, but i had to use a ribbon connector i found and design my own board.