dekuNukem/Nintendo_Switch_Reverse_Engineering #143: Do you know anything about switch game cards? I'm trying to use reverse engineering in one

Arthurfogo7, 2023-04-27:


german77, 2023-05-18:
What game cards? NFC and amiibos? Currently I know how to read them but I've trying to write some data. I almost have the protocol done just need to do some final touches

Arthurfogo7, 2023-05-18:
Yeah, I want to know how to read them, since I think the content is encrypted. Can you teach me briefly?

german77, 2023-05-20:
Yeah most of the content is encrypted. Only a few data like the model_info is in plain format. For encryption see https://github.com/yuzu-emu/yuzu/blob/master/src/core/hle/service/nfc/common/amiibo_crypto.cpp

For reading the raw data from a joycon it's a bit more complicated but at the end of the day you just need to send a few commands to enable it and wait until a tag is in range. This is done here
https://github.com/yuzu-emu/yuzu/blob/master/src/input_common/helpers/joycon_protocol/nfc.cpp

Here is a brief dump of the commands used on the switch while reading an amiibo tag
```
NFC INITIALIZE
[ts:  58706ms t: mc::BtdrvMitmThread    p: 17/17] Report Mode NFC_IR_MODE_60HZ               -> Request: [0x01] 01 0a 00 00 00 00 00 00 00 00 03 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58767ms t: mc::HidReportThread    p: 17/17] <- HID Command Response [0x03]: 21 40 80 00 00 00 12 48 7c f6 47 80 0a 80 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58802ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 0b 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58832ms t: mc::BtdrvMitmThread    p: 17/17] Set MCU state [0x01]                       -> Request: [0x01] 01 0c 00 01 40 40 00 01 40 40 22 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58888ms t: mc::HidReportThread    p: 17/17] <- HID Command Response [0x22]: 21 58 80 00 00 00 12 38 7c f8 37 80 00 80 22 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58940ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 0e 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  58987ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 0f 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  59035ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  59083ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 01 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  59112ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 02 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  60188ms t: mc::BtdrvMitmThread    p: 17/17] Set MCU config [0x21],[0x00],[0x04]        -> Request: [0x01] 01 07 00 00 00 00 00 00 00 00 21 21 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a2 
[ts:  60270ms t: mc::HidReportThread    p: 17/17] <- HID Command Response [0x21]: 21 69 80 00 00 00 10 48 7c ff 37 80 00 a0 21 01 00 04 00 03 00 05 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 51 
INITIALIZATION DONE
[ts:  60289ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 09 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60338ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 0a 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60408ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 0b 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60456ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 0c 00 01 40 40 00 01 40 40 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60514ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 0d 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60557ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 0e 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
[ts:  60751ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand ReadDeviceMode [0x02]   -> Request: [0x11] 11 02 00 00 00 00 00 00 00 00 02 04 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 87 
Polling END
[... cut ...]
[ts:  63524ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 09 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[ts:  63559ms t: mc::BtdrvMitmThread    p: 17/17] Unknown mcucommand SetDeviceMode [0x01]    -> Request: [0x11] 11 0a 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

Arthurfogo7, 2023-05-23:
Thank you for the info. I have a last question, I don't know if you will understand what I mean, but do you know if it is possible to extract data direct from the NAND chip inside the cartridge? With a adapter or something

german77, 2023-05-23:
Oh lol I completely missed the topic. Game cards as game cartridge. Suddenly the topic makes more sense. I was talking about joycons since most RE here is for that.

Cartridges are encrypted with a special key different than game encryption that prevents from reading the contents properly. So far the only reader is the switch as I never heard of an external adapter before. Sadly I don't know much about this topic.