dekuNukem/Nintendo_Switch_Reverse_Engineering #152: Joycon not connecting to Switch, powering up or charging/PCB issue?

scutunic, 2024-02-14:
I am trying to detect an issue on right Joycon but so far no success.
I've seen some videos where fuses are faulty, this was checked and they are OK.
I also have one good right joycon for comparison and I found one difference so maybe there is someone here that could know more about this point on the picture.
[image]
On a working joycon i have 1.8V between GND and this TP and on this faulty one I have zero.
So does anyone know what is this TP related to so I could check further.
Thank you.

ZekiGr95, 2024-04-18:
Hi @scutunic I have the same issue. Not powering on, no "charging" not recognized in the rail. If i connect the rail and attach to the Switch, power is coming into the board and charging the battery.

I'm missing the 1.8V onto the COL (same pad as your) and if you flip the board and check the IC in the back it's a W something there should be also 1.8V but there's none except on a capacitor.

Have you managed to fix yours?

Thanks!

scutunic, 2024-04-18:
> <dekuNukem/Nintendo_Switch_Reverse_Engineering/issues/152/2063707268@
> github.com>

Hi,

Thank you for your comment.

No, I haven't been able to repair it so far. I am self learned and this is
where my knowledge ends :)

DacoTaco, 2024-08-24:
fyi @scutunic and @ZekiGr95 , i had the same issue because i was fucking around in the joycon with the battery attached and i mustve connected some stuff that i shouldnt have. 
apparently there is a fuse/1ohm resistor in the joycon near the battery connector that can blow.
check them to verify.
on the right joycon they are found like in the picture below.

[image]

source : https://www.ifixit.com/Answers/View/636373/Right+joycon+not+charging+after+analogue+stick+change?permalink=answer-681617#answer681617
thanks hexkyz for pointing that out :D