dekuNukem/Nintendo_Switch_Reverse_Engineering #156: Help with Joycon Issue

GustaKir, 2024-07-17:
So, I have some third party joycon that I bought, and apparently both Left and Right Joycon are being detected as the same? input, A friend looked into it and apparently it has to do with  byte 3 instead of 5 and reporting as a right joycon for the subcommand 0x02. Can I fix those issue by myself by inserting some sort of codes or something into the joycon?

WindowStations, 2024-07-27:
Maybe you could use the product id of the device to identify, but I'm not familiar with the source code here to implement that.  I am storing the product id in a UDT, so I can branch them separately when needed.

WindowStations, 2024-07-27:
Byte 13 from command 0x02 has left/right information too.  If the left nibble is non-zero then it is a left Joy-Con.  If the right nibble is non-zero then it is a right Joy-Con.  That is what I'm seeing consistently through older native code.